MallocArray/airgradient_esphome #1: When more the one airgradient installed, only one is providing data

Impatience71, 2022-02-15:
First of all - Thanks for your awsome integration !
When I install 2 airgradients (name is different in the config of course), the ESPHome log shows that both are working, each have different IP addresses, both deliver different values. But in Grafana only one of the sensor provides accessable data. Would it be possible that they collide somehow ?
Peter

MallocArray, 2022-02-15:
I only have one item connected currently, so I haven't tested 2 at the same time, but I suspect it is related to the ID being the same.  On one of the devices, change the sensor IDs to have a unique name, so instead of
```yaml
id: pm2
```
change one of them to something like:
```yaml
id: pm2_basement
```
Then you'll also need to change the references in the display area to also have the new line
```yaml
     - id: display_pm2
        lambda: |-
          it.print(0, 0, id(font1), "PM2");
          it.printf(64, 24, id(font1), TextAlign::TOP_RIGHT, "%.0f",id(pm2).state);
``` 
becomes
```yaml
     - id: display_pm2
        lambda: |-
          it.print(0, 0, id(font1), "PM2");
          it.printf(64, 24, id(font1), TextAlign::TOP_RIGHT, "%.0f",id(pm2_basement).state);
```

Then push the new config in ESPHome and see if the unique values become available.

Impatience71, 2022-02-15:
I will try that out. I assume also that there is something with double identities … 

 

Thanks for the hint !

 

Cheers Peter

Motya-HA, 2023-09-25:
This is absolutely awesome integration. I spent about 2 days trying to make this thing work, but with your code (and change  it to the SHT40) it flow-less.

Just one question - why did you change the fonts between different screens? :-)
I believe this device is screaming for this

https://www.lilygo.cc/products/t-eth-lite

Having ESP32 with POE is the way to go

Motya-HA, 2023-09-25:
BTW, I am using Pro Presoldered-Version, PCB Version 4.2
Any updates for it?

Vladi

MallocArray, 2023-09-25:
@Motya-HA 
I've been trying out different layouts on the screen. I kinda like one compact page that shows almost all sensor data, but it can be hard to read from a distance.  Different fonts at different sizes show up oddly on this limited display, so I was using different fonts for different size text to try to make it look the best I could get it.  Honestly, messing around with fonts took me 2-3 times as long as getting sensors going.

I just got a 4.2 device and I have code that should work for the config button on the back, but I'm also exposing the same button as a switch in HomeAssistant, so you can toggle between F and C for temperature either with the physical button or in HA. I've only had it for 2 days, but I'll have some updated config files ready soon.  Watch or Star this project to get an alert when new stuff is pushed, or I'll try to remember to comment here.

Motya-HA, 2023-09-25:
I am planning to install many of these devices in my future house (I
checked pretty much everything, airthings, Vindstryka with ESP32, airaware
[they are horrible] )
For now this is the best device that includes pretty much everything I
need for air quality

Vladi

geerlingguy, 2023-11-12:
@MallocArray Just chiming in to say thanks for maintaining all these configs!

MallocArray, 2023-12-28:
With the addition of `name_add_mac_suffix: true` to the default config, multiple devices will automatically get unique entries created.

closed 2023-12-28 by MallocArray

Motya-HA, 2024-01-15:
Just looked at your 1.0.0 
Great job...really awesome.
I think I already mentioned it somewhere - instead of mac_suffix I believe the identification as "room" will be much more convenient. Right now it is very difficult to identify entities of different AG devices with mac_suffix. If it will be something like AG_Master_bedroom_PM2.5 (or something like that) it will help tremendously

MallocArray, 2024-01-16:
2.0.0 just released, so check it out as well.  A few things to manually updated, listed as Breaking Changes