MallocArray/airgradient_esphome #6: Particulate measurements stop updating after 1-3 hours, corrected with I2C bus frequency change to 100khz

vshie, 2023-12-07:
Title says important details. Setup newly delivered, pre-soldered air gradient hardware (11/2023.) Use your great yaml, had no issues initially - just lowered the brightness of those damn LEDs. Maybe this contributed to I2C issue, but default of 200-400khz (?) didn't lead to reliable performance. Changed to 100khz based on a very old forum post, 5 days without data interruption since. Sharing for community awareness!

MallocArray, 2023-12-10:
What were you seeing when it was not reliable?  Rebooting or getting values that weren't reasonable?

The increased i2c speed was to address some warning messages about things taking too long and I haven't seen any instability with 400, but may consider lowering it if I can find other items not working correctly.

Which model AirGradient did you use?

vshie, 2023-12-11:
Hi!
I was seeing, and still encounter occasionally, that the particulate sensors stop reporting - temperature and CO2 continue. 
I'm using brand new airgradient one hardware, pre-soldered. 
Power cycling is currently only fix to get particulate data to resume updating...

vshie, 2023-12-11:
https://pastebin.com/5yZc5iTW

MallocArray, 2023-12-12:
Very curious, especially considering the PMS5003 sensor uses UART and not I2C at all, so surprised that it would stop reporting while others work, since it isn't communicating over that method.  I believe you though.

vshie, 2023-12-19:
You were correct, my issue had nothing to do wih I2C! Returned to setting in your code, and removed all airgradient http data transactions - this solved my issue of serial data from PSM5003 stopping after some number of hours.

closed 2023-12-19 by vshie