MallocArray/airgradient_esphome #7: Outdoor monitor fails to get serial number

sbadger, 2023-12-10:
Finally got my indoor and outdoor air monitors and immediately  flashed them with ESP Home. The indoor air unit seems to work well and I was able to pair it with airgraidient.com, but the outdoor unit fails. I'm not able to get the serial number from the log like I did on the indoor unit. The log is showing this error:

```
[07:12:41][C][sgp4x:305]: SGP4x:
[07:12:41][C][sgp4x:306]:   Address: 0x59
[07:12:41][C][sgp4x:307]:   store_baseline: 1
[07:12:41][W][sgp4x:315]: Get Serial number failed.
[07:12:41][C][sgp4x:330]:   Update Interval: 60.0s
```

The network capture shows IT is reaching out but gets a 400 return
```
POST /sensors/airgradient:dc5475bdd200/measures HTTP/1.1
Host: hw.airgradient.com
User-Agent: ESPHome
Connection: keep-alive
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Content-Type: application/json
Content-Length: 156

{"wifi":"-67.000000","pm01":"nan","pm02":"nan","nox_index":"nan","pm10":"nan","pm003_count":"nan","rco2":"nan","atmp":"nan","rhum":"nan","tvoc_index":"nan"}

HTTP/1.1 400 Bad Request
Server: nginx
Date: Sun, 10 Dec 2023 14:08:05 GMT
Content-Type: text/html; charset=utf-8
Content-Length: 51
Connection: keep-alive
X-DNS-Prefetch-Control: off
X-Frame-Options: SAMEORIGIN
Strict-Transport-Security: max-age=15552000; includeSubDomains
X-Download-Options: noopen
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
ETag: W/"33-H/MnJ1LEJ9aArG3Lg/irpV9EDts"

sensor 'airgradient:dc5475bdd200' is not configured
```

MallocArray, 2023-12-10:
Is your outdoor model the one with one PM2.5 sensor and one CO2 sensor?

The sgp sensor it is complaining about is only on that model. If you have the one with 2 PM 2.5 sensors, you'll need a different config, but I haven't built one yet.

Another user has contributed one, I can try to find the link later today

sbadger, 2023-12-10:
Yes, I have to one with no CO2 and 2 PM 2.5 modules. I will wait for the new config. Thanks!

sbadger, 2023-12-10:
It turns out their are 2 versions of the outdoor sensor!

closed 2023-12-10 by sbadger