MallocArray/airgradient_esphome #8: Airgradient One (v9) LEDs Keep Turning Back On

msirianni, 2023-12-19:
Flashed a new pre-assembled kit with ESPHome via your YAML and it is mostly working, but when I turn off the LED strip via HA it initially works, but then turns itself back on a few seconds later. This repeats every time I turn it off via HomeAssistant. 

Separately, it seems the LEDs are all constantly on and only Green no matter the detected AQI. Is there a way to restore the behavior of the stock LEDs where they move from left to right and change color to match AQI conditions? 

Thanks!

MallocArray, 2023-12-19:
The LEDs always being on is a current limitation, as each time the CO2 sensor gets a value it turns the lights back on to the appropriate value.

When I first got my V9 kit, the LEDs were associated with the CO2 values. It will be green until it goes over 800 then start going orange, red, purple, etc.

I am in the middle of updating this so the brightness can be set in HA and will be persistent so you can make it dimmer or off entirely.

If AirGradient changed the LED to reflect the AQI, that would take a bit more work to add a config for, but could be done.  I don't think the AQI values are that accurate for indoor, so it is a lower priority for me right now.

I'm maybe a day away from some new BIN files and hopefully I'll have the yaml files ready to share at that point as well.  I still have the outdoor monitor to test with my updates.  I'll let you know.

MallocArray, 2023-12-19:
I just uploaded the latest .bin files to the main branch if you want to manually upload the latest.

In the `includes` branch, I've started splitting out the different components so I can refer to one file across different models without having to maintain the configs individually.  It may be a bit confusing while I finish working on it and come up with some resulting files that have all of the separate ones combined into standalone files, but that will come later.

strikeir13, 2023-12-19:
As a temporary measure, I adjusted the One V9 esphome config so that my Home Assistant input boolean 'Quiet Hours' will override the LED updates and keep the LEDs off when I turn them off at night. Here's all I changed:

Pulled in the Home Assistant input boolean entity:
```
text_sensor:
  - platform: homeassistant
    name: "Quiet Hours - Home Assistant"
    id: quiethours
    entity_id: input_boolean.quiet_hours
```

Added a check on the above sensor within each of the CO2 condition lambdas, like this:
```
      on_value:
        - if:
            condition:
                lambda: 'return id(co2).state < 800 && !id(quiethours).state;'
            then:
              - light.turn_on:
                  id: led_strip
                  brightness: 100%
                  red: 0%
                  green: 100%
                  blue: 0%
        - if:
            condition:
                lambda: 'return id(co2).state >= 800 && id(co2).state < 1000 && !id(quiethours).state;'
            then:
              - light.turn_on:
                  id: led_strip
                  brightness: 100%
                  red: 100%
                  green: 100%
                  blue: 0%
```

Just make sure to add the `&& !id(quiethours).state` section to each CO2 condition lambda. After these changes, you can turn on the Quiet Hours switch in Home Assistant, turn off the Airgradient LED strip, and it will stay off until you turn off Quiet Hours. At that point, it will turn the lights on at the next sensor update.

Motya-HA, 2023-12-19:
Having LED "on" all the time is a non starter for me - I am placing sensors in the bedrooms for air quality control. I think the following options can be helpful:
1. Turn LED completely off
2. Turn LED off during night (from time "X" to time "Y"). Better option is if it can be controlled by Home assistant device input - I have "home status" virtual device with "home", "sleep", "away" and vacation states. So, turn off LED during "sleep" state will be handy
3. Turn on LED only when it is "red" (air quality is bad), and wait for "turn off" command

MallocArray, 2023-12-19:
> Having LED "on" all the time is a non starter for me - I am placing sensors in the bedrooms for air quality control.

Agreed, and it is being addressed.  My original attempt was to match the AirGradient Arduino code which also had them on all the time.

In the latest compiled .bin in the main branch I now include a switch to turn the LED off and on as well as brightness, so a HA automation could be set to modify the Brightness to 0 during your quiet hours and then return it to your desired brightness for normal times.

And since all of the YAML files are freely available, you could remove the LED colors for anything below the Red threshold if you only wanted lights when it crosses into the Red territory.

Right now I'm thinking of letting HA be the one that controls when to turn LEDs off an on, as I think it would get pretty complicated pretty quick to do it all directly inside of ESPHome.

strikeir13, 2023-12-19:
> > Just make sure to add the `&& !id(quiethours).state` section to each CO2 condition lambda. After these changes, you can turn on the Quiet Hours switch in Home Assistant, turn off the Airgradient LED strip, and it will stay off until you turn off Quiet Hours.
> 
> This is nearly what I did, only I set an `if` at the top that checked if the switch was turned off for the LEDs, then turn off the lights, else, do all of the other regular code I had. This prevented needing to add the condition to each CO2 level https://github.com/MallocArray/airgradient_esphome/blob/includes/includes/sensor_s8_led.yaml
> 
> But then later I also added a Brightness slider that adjust the LED brightness. Now you could set the brightness to 0 and it should result in the same outcome, so my switch for LED off or on may not be needed at all.

This is a much better solution with the top-level `if`. Thanks for this and for the full esphome config!

Motya-HA, 2023-12-19:
> And since all of the YAML files are freely available, you could remove the LED colors for anything below the Red threshold if you only wanted lights when it crosses into the Red territory
Ohh...so I actually just "comment" it in YAMAL code?

MallocArray, 2023-12-19:
> Ohh...so I actually can just "comment" it in YAMAL code?

You could, or just delete the affected lines.
Or better yet, instead of each `then:` setting the light on, you could force it to turn off

```yaml
                    - light.turn_on:
                        id: led_strip
```

That way, if it does need to turn on the red and then the condition clears, unless you have somewhere to turn off the LED, they will just stay on

MallocArray, 2023-12-21:
I just uploaded new config files that has the most recent changes to the main branch.  This will no longer keep turning the LEDs back on and also let you adjust the brightness, among other improvements.

closed 2023-12-21 by MallocArray