MallocArray/airgradient_esphome #11: AG One v9 - LED strip flashes once on boot regardless of light and number restore options

strikeir13, 2023-12-22:
I tried a number of different options for keeping the LED strip off at boot, like initializing the slider at value=0 or setting the light restore to always off, but I've been unable to keep the LED strip off when the device boots. From what I can tell in the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf), GPIO10/pin 16 isn't pulled high at boot, which is what makes me think there is something in the yaml code that is causing the LEDs to flash once at boot. 

Is the LED flashing once at boot a hardware limitation, or is there a yaml configuration that would keep the LEDs off until the CO2 code activates them?

MallocArray, 2023-12-22:
You are using the latest yaml file from the main branch? I'm not experiencing that with mine that I can tell.  Originally I had it stay dark until it got a value from the CO2 sensor, but with the latest one I had it persist with the last status, so if it was on and orange, it should restore to the same and then adjust on the first reading.

Make sure you are using the latest config file and let me know.

Only when you first plug it in, or are you talking every time you upload a new config that does a soft reset as well?

strikeir13, 2023-12-22:
I pulled the last yaml from the main branch and did some more testing. My use case is that this sensor is in a bedroom, so I'm trying to ensure that the LEDs and screen stay off when I want them to stay off. To that end, I've made a few tweaks to the latest yaml:

- I'm using some display code from the airgradient forum to mimic the original arduino display. I also have a blank page and the bootup page from the latest yaml.
- I pulled in a Home Assistant boolean to reference in the display automation code and that is working well.
- I haven't made any changes to the light or number code other than testing `restore_mode: ALWAYS_OFF`

Using the latest yaml, there are 4 LEDs in the strip that blink on momentarily at boot, for both soft resets (e.g. uploading new code) or power cycles (e.g. unplugging). Since it's not all LEDs in the strip, my guess is that this is probably a hardware behavior and that no amount of automation or `restore_mode:` changes will make a difference.

To reproduce the behavior, I either turn the LEDs off manually or set `restore_mode:` to `ALWAYS_OFF`, then unplug/plug in the device. There is a brief blink of some LEDs in the strip before they stay off until the CO2 code turns them on.

MallocArray, 2023-12-22:
I'll see if I can reproduce after the holidays.  I also plan on adding the default display layout in the next round of updates.

If you are willing, could you try flashing the .bin file with the ESPHome web flash tool and see if the flashing LEDs still happens with it?  You can then go back to your current config, but there could be a slight difference between the compiled .bin and the latest yaml, and I'm running the .bin on mine.  If it behaves differently, then I've messed something up in the yaml

strikeir13, 2024-01-14:
Closing this issue - I've been doing some testing today and I haven't been able to reproduce it again.

closed 2024-01-14 by strikeir13