MallocArray/airgradient_esphome #12: LED strip turns off after about a minute

smithdo, 2023-12-31:
Using your 0.2.2 firmware and one of the devices the LED strip turns off after about a minute at most.  Any ideas?

MallocArray, 2023-12-31:
How long does it stay off?

Any other changes to the config file?  If the watchdog code was removed, it would reboot after about 2 minutes.

I have some newer changes coming in a few days. I'll try to remember to reply here

If you are connecting to Home Assistant, check the Uptime sensor to see if the device is rebooting (indicated by the Uptime changing to a low value)

smithdo, 2024-01-01:
I turn it on. It turns off between 30-60 seconds and stays off until I turn it back on.  Reloaded esphome firmware.  Rebooted.   I do have the same config (different substitutions) and works fine on one and does this on the other.  Logs don’t say why, just that it turns off.

MallocArray, 2024-01-01:
That is very odd, especially if different devices are behaving differently.  Did you check the Uptime values to see if it looks like the problematic device is rebooting?  That shouldn't keep the LED off, only until the CO2 sensor gets a reading.

Does the display show the boot screen often, where it shows the serial number and config version number? That would also indicate it is rebooting.

smithdo, 2024-01-01:
Fixed it, did a clean build files and then recompiled and deployed and it is now staying on.

smithdo, 2024-01-02:
Less than 12 hours and it went back to the same problem.

MallocArray, 2024-01-02:
I released a significant update yesterday to this repo.  Hopefully the Readme covers the majority of what changed and what you may need to change.

Give it a shot and see if it is more consistent.  If the device is rebooting, this latest update should keep the LEDs on during the startup, so they shouldn't flash off.

Just curious, do you have multiple Wireless Access Points in your home?

pashdown, 2024-01-02:
Not directly related, but should the LEDs relate to CO2 levels or AQI?  I'd think AQI would be more useful.

MallocArray, 2024-01-02:
> should the LEDs relate to CO2 levels or AQI?

Further discussion can be had on the AirGradient forum, but it is something that could be an option, although I wouldn't necessarily use AQI but use PM2.5 instead, which is what AQI is based on. (I personally want CO2, but could very easily see others wanting a PM2.5 based light)

smithdo, 2024-01-02:
The device does not reboot.  I do have multiple APs. I use UniFi.  I think it maybe a hardware issue.  I updated the code and pushed to both devices.   I tested the one with the issue and it still happened.  Turn it on and 33s later it turned off.  So I tried a new thing.  I turned the brightness down to 40% and it stayed on.  I have even raised it to 50% and it is still working for 6 hours now.  Tomorrow I will increment my increase until it happens.

smithdo, 2024-01-04:
Still testing so far so good, up to 80% brightness

MallocArray, 2024-01-16:
Certainly sounds like an issue with current draw, but I've ran mine at full 100% with no issue.  

I just pushed a new update that also bring the ability to change the contrast of the screen so it can be dimmed as well.

closed 2024-01-16 by MallocArray