MallocArray/airgradient_esphome #13: Multiple sensors  friendly entity name

Motya-HA, 2024-01-01:
Hello
I know that Malloc already created solution for multiple sensors with  the following line:
name_add_mac_suffix: true  # Set to false if you don't want part of the MAC address in the name

In that case entity is looking like that:
"airgradient_one_0c0acc_pm_0_3" , where "0c0acc" is a MAC address

I am wondering if there is a way to create aenetety that reflects location, for example: for "master bedroom" - "airgradient_one_master bedroom_pm_0_3", for "guest room" :"airgradient_one_guest_room_pm_0_3" etc.

Thanks

MallocArray, 2024-01-01:
You could change the `devicename: ` field to be whatever you wanted it to be named, and if you don't want the MAC added, set that to false. Just make sure the device name is unique to all of the device in your HomeAssistant

So example would be 
```yaml
devicename: "airgradient_one_master_bedroom"
upper_devicename: "AG One Master Bedroom"
add_mac_suffix: "false"
```

Motya-HA, 2024-01-01:
Hmmm
I changed it like that:
substitutions:
devicename: "airgradient_one_master_bedroom"
upper_devicename: "AG One Master Bedroom"
ag_esphome_config_version: 0.9.10

esphome:
name: "${devicename}"
friendly_name: "${upper_devicename}"
name_add_mac_suffix: false # Set to false if you don't want part of the MAC
address in the name

However, it is still : "*airgradient_pro_1_pm_2_5*". I had an ID: "1"
before. ESPHome did not change it

Probably I should removed device completely and re-ender it

MallocArray, 2024-01-01:
I agree that you may need to remove it from HA and add it back in.

Do make sure that after you changed your YAML file, that you did the Install process again to send the latest code to your AirGradient.

Motya-HA, 2024-01-01:
"Do make sure that after you changed your YAML file, that you did the
Install process again to send the latest code to your AirGradient" - I know
that much :-)

If I am doing that I'll do the "includes" version. I am a bit confused, do
I need to copy all files from the "include" folder or from the  "packages"
folder?

And copy the it under the folder "packages" in

   - /homeassistant/espho

MallocArray, 2024-01-01:
At this point you only need the packages folder in addition to the top level file for the particular model.

When I get a few more minutes I'll update the readme more and I have a more straight forward way to address it as well

MallocArray, 2024-01-02:
Did a large update in the main branch last night, so please tke a look

Motya-HA, 2024-01-03:
Hi Malloc
If it takes the packages from github it means that we can't fine tune the code, correct. For example, I wanted to change moving average for PM that won't be possible.
Am I missing something?

MallocArray, 2024-01-03:
Still have options available.  The simplest is just using the single file that references the packages in the GitHub, but if you want to modify it yourself, you can copy the individual package from the repo, such as the https://github.com/MallocArray/airgradient_esphome/blob/main/packages/sensor_pms5003.yaml file and put it in your own ESPHome config folder (or a subfolder named packages for consistency) and then change the line to 
`pm_2.5: !include packages/sensor_pms5003.yaml` and now it will use the file you have saved locally and you can make whatever modifications you would like, while all of the other packages continue to come directly from Github.  So it can be flexible.

The default package does a sliding moving average over 30 seconds with 1 second updates, or you could change it to `sensor_pms5003_extended_life.yaml` which does a single reading every 2 minutes with no averaging, or use the above to modify either file as you see fit.  

The only downside if that if the repo changes to update something in a file you've copied locally, you would have to merge in any changes you wanted yourself.

MallocArray, 2024-01-16:
I pushed version 2.0.0 tonight and the README.md file now contains additional information about how to either copy an individual package file locally and update your file to reference it, or use the Extend feature to modify certain parts of an existing package without needing to copy them locally.

closed 2024-01-16 by MallocArray