MallocArray/airgradient_esphome #14: Home Assistant - Airgradient ONE does not become online

ericmuijs, 2024-01-05:
1. I created a new device via the configuration of ESP Home Addon and copied the wifi and ap settings.
2. Copied the yaml code of the Airgradient One device
3. Pasted in the settings of 1
4. ESPHome does not list the device as online, API code cannot be found also when clicking the menu and "show API key"

MallocArray, 2024-01-05:
If you only have a single device, try changing this line to false
`  add_mac_suffix: "false" `

when it adds the MAC to the device name, then ESPHome doesn't know what the full name ended up being, so it will probably show it as Offline, even if it actually is online, since it can't ping or discover it by name.

In my configs I also add a static IP config, and I have my ESPHome set to ping by IP, so it can correctly find if it is online or not, but removing the MAC suffix is a more straight forward way if you don't have multiple of the same model in your environment.

Static IP would look something like this:
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: 192.168.1.85
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 4.2.2.2
```

Schlats, 2024-01-07:
I had the same issue, just changing the mac_suffix from true to false got it working for me. Static IP is a good idea, will incorporate in mine as well now that I can do OTA updates.
Thanks,

MallocArray, 2024-01-16:
I just pushed version 2.0.0 of my config and one of the changes is to default that to false, so it will behave more reliably.

closed 2024-01-16 by MallocArray