MallocArray/airgradient_esphome #16: Difference between installing the .bin and .yaml installs on AG One?

Schlats, 2024-01-07:
I first installed the AGone.bin, HA found it there were many options, but didn't know how to then get it to show up in ESPhome admin addon. I then installed the AGone.yaml via the ESPhome admin addon, think it was the same as the .bin. I found it is different, doesn't have all the OLED window options. How can I get all the options of the .bin in the .yaml?

Appreciate the work into this, as I barely have an idea what I'm doing, but enjoying learning this stuff.

Thanks,

MallocArray, 2024-01-07:
The file airgradient-one.yaml should be identical to airgradient-one.bin as I used the YAML to create the .BIN

Once it is adopted to HomeAssistant, you should see all of the different display options for the OLED as switches you can turn off an on, but by default it will only show the single summary page, but the .bin should have behaved the same.
[image]

If you aren't using HomeAssistant, but only ESPHome, then you will need to enable the web server by adding this line to your main file somewhere:
`web_server:`

Then once you install the updated configuration, you should be able to open a web browser to the IP of your AirGradient and be able to toggle the switches for the different display pages

P.S. If you are having any other issues discovering your device after install, try changing this line to false so it doesn't add the MAC to the device name
`  add_mac_suffix: "false"`

Schlats, 2024-01-07:
I just went through trying uploading the .bin to show the difference, but I can't seem to get it to do it again. I don't recall if the section of switches was called Configuration, but via the esphome integration the AG-One device had switches for the OLED like you have shown, but I thought there were more. From just building it via the ESPHome admin panel in HA I no longer have that section and OLED just scrolls through multiple displays. The default AG screen suits me fine. I only have 2 items in the configuration section. 

I've also noticed the LED slider doesn't work, but if I click on the LED for color, the slider in there works.

MallocArray, 2024-01-07:
If you aren't running ESPHome 2023.12.5, you might try upgrading to that, since that is what I built the .bin with.

Schlats, 2024-01-07:
I've been on 2023.12.5 for a bit now, so it was definitely done with it.

I realized I forgot to connect to the AP hotspot to setup wifi when installing the .bin hahaha. Anyhow I installed .bin, gave it wifi access, HA gave notification for esphome device found, I configured it via esphome integration, then via ESPHome addon I added new device, copy pasted .yaml and bam now have 29 entities under AG device all is good.

It seems weird if it is built via ESPHome addon with .yaml, then download .bin and install on device, I would only get 22 entities under the AG device.

Thanks for the help and quick responses, much appreciated.

Schlats, 2024-01-07:
forgot to close with comment

Schlats, 2024-01-08:
Okay, was curious about the ```web_server: implentation so added it to the yaml via ESPhome addon and now the OLED switches disappeared. Can see them grayed out below.

[image]

Schlats, 2024-01-08:
Oh goodness, no sure how but apparently I had downloaded ag_esphome_config_version: 0.2.1 .yaml. I think I have it figured out now. Used the ag_esphome_config_version: 1.0 yaml and the switches are back.

closed 2024-01-08 by Schlats