MallocArray/airgradient_esphome #17: AG Open Air O-1PST not working.

coltondick, 2024-01-11:
I've been trying to get this to work on my AG Open Air O-1PST, but I haven't had any success. 

I received my kit, which appears to be slightly different from the current kit, since it only included 2 x PMS5003T. 

I've commented out the other sensors and the Air Gradient API from my configuration and flashed it using esphome. I've been able to add the sensor to my Home Assistant instance, but all of the values are displayed as "Unknown." 

[image]

Any assistance with this would be greatly appreciated.

MallocArray, 2024-01-11:
I have all of the parts to make the version with 2 PM sensors, but haven't gotten around to it yet. You are seeing Uptime and Wifi so that much is working, but I'm surprised that none of the sensors are returning anything.

You might give this a shot, note that I changed add_mac_suffix to false so it may need to be re-imported as a new entry:

```yaml
substitutions:
  devicename: "ag-open-air"
  friendly_devicename: "AG Open Air"
  ag_esphome_config_version: 1.0.0
  add_mac_suffix: "false"  # Must have quotes around value

# Enable Home Assistant API
api:  # Add encryption key as desired

ota:  # Add password as desired

wifi:
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
  safe_mode: github://MallocArray/airgradient_esphome/packages/switch_safe_mode.yaml
```

You can also try this config from another AirGradient fan that is built for the same model you have.  I plan on using this to get the averaging figured out.
https://github.com/ajfriesen/ESPHome-AirGradient/blob/main/air-gradient-open-air.yaml

MallocArray, 2024-01-18:
I have the initial attempt at the O-PPT1 now published in a separate branch.  If you would like to try it out, you can get the main fil here:

https://github.com/MallocArray/airgradient_esphome/blob/o-ppt1/airgradient-open-air-o-1ppt.yaml

closed 2024-01-22 by MallocArray