MallocArray/airgradient_esphome #18: AG Open Air O-1PP on a 1-PST board?

jayofdoom, 2024-01-13:
I have a strange-seeming AG Open Air O-1PP. It's essentially the O-1PP as listed on their website here: https://www.airgradient.com/documentation/diy-open-air-presoldered-v11/ EXCEPT the board is clearly the one pictured for the O-1PST: https://www.airgradient.com/images/pages/open-air/oa_board_front_90.webp

I know this is outside of your current support window entirely; but I'm willing to assist with information in getting this supported if it's something you're interested in. 

For now, I'm testing some of the O-1PP targeted configurations from AJFriesen.

MallocArray, 2024-01-16:
Thanks for reaching out.  Adding support for the dual PM2.5 sensors is on my roadmap and I'll let you know when I have something to try, since I don't have one physically in my hands, only the one with PM2.5 and CO2

seelenamt, 2024-01-17:
I have this too, actually. Ordered just before the new year and received in the past couple days. I'm willing to provide any info I can also.

MallocArray, 2024-01-18:
I have the initial attempt at the O-PPT1 now published in a separate branch.  If you would like to try it out, you can get the main fil here:

https://github.com/MallocArray/airgradient_esphome/blob/o-ppt1/airgradient-open-air-o-1ppt.yaml

@seelenamt @jayofdoom

closed 2024-01-22 by MallocArray

jayofdoom, 2024-01-23:
This has been running on mine for approximately 30 minutes. Seems to work. I did observe a "400 bad request" error trying to upload stats to AirGradient:

`[01:03:43][W][http_request:093]: HTTP Request failed; URL: http://hw.airgradient.com/sensors/airgradient:redacted/measures; Code: 400; Duration: 1067 ms`

Other than that, anything that should be reporting at this point, is.

MallocArray, 2024-01-23:
@jayofdoom If you only saw the Code: 400 once, then things are probably ok, especially if it was during the initial boot.

If you see it consistently, do let me know. If some of the sensors were not initialized, maybe it didn't have good data to send right away and then got proper values.