MallocArray/airgradient_esphome #19: NAN for temp and humidity?

zestysoft, 2024-01-15:
Hi, I've got an AirGradient DIY Air Quality Sensor (Pro Version PCB Version 3.1).
With the original firmware, temp, humidity, co2, and pm2.5 values were displayed on the oled panel and were uploaded to the airgradient dashboard.

I thought I'd give your firmware a try since I want to integrate this with HA, however after uploading the pro bin file via the web method and after setting the wifi details via the device's own access point ssid (non yaml setup), the result is a screen that shows co2 and pm2.5, but not the temp or humidity.

Oddly, my instance of esphome doesn't automatically detect the device as it usually does when I bring up an esphome device?  I was able to add the unit to HA, and it too only sees CO2 and PM2.5 values so I thought I'd ask you if you have any idea what could be happening here?  Normally I'd wait for ESPHOME to detect the device and then properly set it up with the yaml details.

zestysoft, 2024-01-15:
Update -- I wiped the unit and installed the vanilla esphome firmware first -- that let it get detected by my instance of esphome where I could then push the yaml file to it.  So I've at least solved that part of this and I can readily edit and push updated yaml configs to the device now.

However, the original problem still remains -- no temp or humidity values.  If it helps, the diy docs are here: https://www.airgradient.com/documentation/diy-pro/

I believe the esp devices is an 8266, so I'm wondering if maybe newer versions of this hardware have since moved to esp32 and there could be some kinda pinout issue.

MallocArray, 2024-01-16:
The issue with it not being automatically detected in ESPHome is due to the fact the config had the option to add the MAC address as the suffix to the device name.  This caused ESPHome to look for just the base name, such as "ag-pro" but the device was named something like "ag-pro-abc1234" and so ESPHome could not find it.

I just pushed version 2.0.0 of my config and one of the changes is to default that to false, so it will behave more reliably.

I think your issue with no temp/humidity is that your device likely came with SHT31 instead of the SHT41 that the later Pro versions had and is what is in the config.

Try updating your config to the latest 2.0.0 (Note, the name of the substitution vars were all changed, but same purpose)

In the line under packages for temp_humidity, change it from ending with `sht40.yaml` to `sht30.yaml` and then install it again, and I suspect that will give the values.

zestysoft, 2024-01-16:
Ah, yes I've got the SHT31.  Thanks for the fix!

As for the esp detection thing, I'm not sure how the name of the device matters? Usually when I introduce a new esphome device to the network, my esphome instance detects this automatically and asks to "adopt" it. I believe it does this through mdns broadcasting.

zestysoft, 2024-01-16:
Here's an example:
[image]

MallocArray, 2024-01-16:
That is the intended result, but when you adopt that device and it import the YAML file, if the name was just set to "athom-garage-door" and then had the field set to add the mac as the suffix, ESPHome is looking to DNS for "atham-garage-door" and the actual device name is "atham-garage-door-d9454e" so they won't match and ESPHome will show it as Offline. 

If you have a static IP defined, then it can still find it through that, even though the names don't match.

Ultimately, you got it working and currently the default is to not include the MAC, although it is user configurable.

closed 2024-01-16 by MallocArray

zestysoft, 2024-01-16:
To be clear, I'm talking about the **initial setup** -- the esphome instance never detected this device at all - it's not the situation where it detected it and it shows as offline because of a name change skew and now it want's to adopt another copy.

I've got plenty of other esphome devices that use the mac address as the suffix.  It's a great way to set up multiple devices without them clobbering on each other (I have two airgradients running now for instance). I'm not sure what removing the suffix is supposed to fix unless someone is manually setting the dns name of the device and not using the same name?  Weird.

Maybe the issue was something else -- like it didn't properly acquire a wifi connection or something.  I did notice that happen after rebooting the device once.

MallocArray, 2024-01-16:
In that case I'm not entirely sure.

I'm looking at adding esp32_improv which will help it be detected by other bluetooth devices to show up in the dashboard, but it consumes 30%+ of the Flash storage and makes it nearly maxed out.  If you have the full YAML of other devices that were auto-detected, can you see if that is present in their configs?

zestysoft, 2024-01-16:
When I couldn't get your firmware to appear, I switched to this one and it came up without an issue: https://github.com/ajfriesen/ESPHome-AirGradient/blob/main/air-gradient-pro-diy.yaml

Maybe that might shed some light on this?  Otherwise, I have some extra esp32's -- I could do some further testing to see if maybe it was just the unit coincidentally having trouble acquiring wifi.