MallocArray/airgradient_esphome #23: Unable to compile config with esphome 2023.12.8

robinsmidsrod, 2024-01-22:
I'm getting this error when trying to compile your configuration:

```
INFO ESPHome 2023.12.8
INFO Reading configuration /config/esphome/airgradient.yaml...
WARNING Found '${name}' (see esphome->name) which looks like a substitution, but 'name' was not declared
WARNING Found '${friendly_name}' (see esphome->friendly_name) which looks like a substitution, but 'friendly_name' was not declared
WARNING Found '${name_add_mac_suffix}' (see esphome->name_add_mac_suffix) which looks like a substitution, but 'name_add_mac_suffix' was not declared
WARNING Found '$config_version' (see esphome->project->version) which looks like a substitution, but 'config_version' was not declared
WARNING Found 'it.printf(0, 0, id(open_sans_14), "ID:");
it.printf(128, 0, id(open_sans_14), TextAlign::TOP_RIGHT, "%s", get_mac_address().c_str());
it.printf(0, 21, id(open_sans_14), "Config Ver: $config_version");
it.printf(0, 42, id(open_sans_14), "$friendly_name");' (see display->0->pages->1->lambda) which looks like a substitution, but 'config_version' was not declared
WARNING Found 'it.printf(0, 0, id(open_sans_14), "ID:");
it.printf(128, 0, id(open_sans_14), TextAlign::TOP_RIGHT, "%s", get_mac_address().c_str());
it.printf(0, 21, id(open_sans_14), "Config Ver: $config_version");
it.printf(0, 42, id(open_sans_14), "$friendly_name");' (see display->0->pages->1->lambda) which looks like a substitution, but 'friendly_name' was not declared
WARNING Found '${name}' (see esphome->name) which looks like a substitution, but 'name' was not declared
WARNING Found '${friendly_name}' (see esphome->friendly_name) which looks like a substitution, but 'friendly_name' was not declared
WARNING Found '${name_add_mac_suffix}' (see esphome->name_add_mac_suffix) which looks like a substitution, but 'name_add_mac_suffix' was not declared
WARNING Found '$config_version' (see esphome->project->version) which looks like a substitution, but 'config_version' was not declared
WARNING Found 'it.printf(0, 0, id(open_sans_14), "ID:");
it.printf(128, 0, id(open_sans_14), TextAlign::TOP_RIGHT, "%s", get_mac_address().c_str());
it.printf(0, 21, id(open_sans_14), "Config Ver: $config_version");
it.printf(0, 42, id(open_sans_14), "$friendly_name");' (see display->0->pages->1->lambda) which looks like a substitution, but 'config_version' was not declared
WARNING Found 'it.printf(0, 0, id(open_sans_14), "ID:");
it.printf(128, 0, id(open_sans_14), TextAlign::TOP_RIGHT, "%s", get_mac_address().c_str());
it.printf(0, 21, id(open_sans_14), "Config Ver: $config_version");
it.printf(0, 42, id(open_sans_14), "$friendly_name");' (see display->0->pages->1->lambda) which looks like a substitution, but 'friendly_name' was not declared
Failed config

esphome: None
  
  '$' is an invalid character for names. Valid characters are: abcdefghijklmnopqrstuvwxyz0123456789-_ (lowercase, no spaces).
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: ${name_add_mac_suffix}
  project: 
    name: mallocarray.airgradient
    version: $config_version
  min_version: 2023.12.0
  on_boot: 
    priority: 200
    then: 
      if: 
        condition: 
          switch.is_on: upload_airgradient
        then: 
          - http_request.post: 
              url: !lambda |-
                return "http://hw.airgradient.com/sensors/airgradient:" + get_mac_address() + "/measures";
              headers: 
                Content-Type: application/json
              json: 
                wifi: !lambda |-
                  return to_string(-50);
```

This is my configuration:

```
# AirGradient ONE - Board v9
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  devicename: "airgradient-one"
  friendly_devicename: "AirGradient One"
  ag_esphome_config_version: 1.0.0
  add_mac_suffix: "false"  # Must have quotes around value

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  temp_humidity: github://MallocArray/airgradient_esphome/packages/sensor_sht40.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  display: github://MallocArray/airgradient_esphome/packages/display_sh1106_multi_page.yaml
  led: github://MallocArray/airgradient_esphome/packages/led.yaml
  led_co2: github://MallocArray/airgradient_esphome/packages/led_co2.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  config_button: github://MallocArray/airgradient_esphome/packages/config_button.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
  safe_mode: github://MallocArray/airgradient_esphome/packages/switch_safe_mode.yaml

binary_sensor:
  - id: !extend config_button
    pin:
      ignore_strapping_warning: true  # Acknowledging that this is a strapping pin and should not have external pullup/down resistors  https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret access_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Airgradient Fallback Hotspot"
    password: !secret wifi_password

captive_portal:
```

MallocArray, 2024-01-22:
I renamed the substitution variables in v2.0.x so they match the ESPHome vars exactly, so you'll need to revise the names.

```yaml
substitutions:
  name: "airgradient-one"
  friendly_name: "AirGradient One"
  config_version: 1.0.0
  name_add_mac_suffix: "false"  # Must have quotes around value
  ```
  
  Also wanted to point out that you have `logger:` in your config, but this is also set in the `packages/airgradient_esp32-c3_board.yaml` file with the buad_rate set to 0 to disable it, since both hardware UARTs are being used by sensors, so it won't work for logging.

robinsmidsrod, 2024-01-22:
Thanks a lot! It validates now. And installing with the latest ESPHome version works as expected.

closed 2024-01-22 by robinsmidsrod