MallocArray/airgradient_esphome #24: Sht4x sensor measurements are off - how to adjust in yaml

bramboe, 2024-01-24:
I have multiple Airgradients in my home. I noticed the measurement of the Sht4x are off compared to other (multiple) temp/hum in the same rooms (by more than 2-3 degrees c.). After doing some searching online I assume it has something to do with the heater configuration. I'd like to either turn it off or add an offset to the setup in ESPhome to test what gives the best results

I've tried to use the '!extend' feature by talking to the 'temp' ID placing it under a 'sensor' in my yaml. But that ID won't be found when I validate my script. Is there any way to make some modifications to the current yaml or this (amazing!) esphome integration?

Thank you for making this integration within ESPhome available!

- Bram

MallocArray, 2024-01-24:
Is it off be reporting too high or too low?

Make sure the AirGradient is mounted vertically, either hanging on a wall or using the little stand feet so there is room for air to enter the vents at the bottom and rise through the top vents. If you have it laying flat, it will cause a discrepancy. 

I'll play with some extend options to enable what you are looking for, or you can always copy the `sensor_sgp41.yaml` file to your local device, make your modifications, and then replace the package line for it with something like (assuming put the file in a subfolder named packages)
```
temp_humidity: !include packages/sensor_sht40.yaml
```

bramboe, 2024-01-24:
He! Thank you for your quick reply! The temp and hum are both too high. My Airgradients are both hanging vertically on the wall with enough space around them. 

Thanks for your advice on the local file modifications, I'll do that for now.

bramboe, 2024-01-26:
Update here: I ended up with adding an offset, both to my humidity and temperature. Now the measurements on my AirGradients match closer the temperature and humidity of my other sensors. 

FYI, tough, all these sensors are near each other and I live in a small apartment the offsets are quite significant. one offset for my humiditity is +10 and +20 and for my temperature is -6.

cheezgr8r, 2024-02-26:
I'm having a similar issue with my SHT4x sensor. I assembled my AG One on Saturday and it's been up since yesterday. I have it on a shelf about 5.5 feet off the ground. There is nothing around the AG that is generating heat and I'm using the included feet as a stand. The AG One is consistently reading 13F higher than I would expect it to. I tried using both the full and condensed yaml's and moved the AG One to different rooms. 

@bramboe, how did you add an offset?

cheezgr8r, 2024-02-29:
After noting a change in temperature reading after turning off the LED's and doing some more digging it turned out that I swapped the SHT4x and VOC sensors during assembly. Temperature and humidity readings are now reading accurately.