MallocArray/airgradient_esphome #25: Sensors not all reporting on AG Open Air O-1PST

chrismcneil, 2024-01-26:
I have a AG Open Air O-1PST and on stock firmware all sensors are reporting but when I switched to https://github.com/MallocArray/airgradient_esphome/blob/main/airgradient-open-air-o-1pst.yaml a number of the sensors no longer report in.

```
# AirGradient Open Air Outdoor Monitor with CO2 and TVOC
# Model: O-1PST
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  name: "ag-open-air-o-1pst"
  friendly_name: "AG Open Air O-1PST"
  config_version: 2.0.2
  name_add_mac_suffix: "false"  # Must have quotes around value

api:
  encryption:
    key: "REDACTED"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-Ba98Ec"
    password: "xp5wHkgv7jRl"

# Turn on the webserver so we can connect and look at raw stats
web_server:

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-open-air-o-1pst.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
  safe_mode: github://MallocArray/airgradient_esphome/packages/switch_safe_mode.yaml
```

[image]

MallocArray, 2024-01-26:
That is odd.  This is a stock O-1PST and not one you swapped sensors into, right?

I'm running this same code on mine and I'm able to see all of the sensor data inside of HomeAssistant and the AirGradient dashboard, I haven't enabled the webserver to see anything but if it isn't showing there, I'm pretty sure it won't be passing it along to HA either.

Can you try removing power for a few seconds and then powering it back up? Some of my indoor sensors occasionally need that to get the PM sensor to fully register

chrismcneil, 2024-01-26:
I ordered from the shop and have left it stock SKU: #open_air_O-1PST

chrismcneil, 2024-01-26:
🤦‍♂️ power cycling seems to have done the trick!

closed 2024-01-26 by chrismcneil