MallocArray/airgradient_esphome #26: PM 0.3 unit

kroeckx, 2024-01-28:
The PM 0.3 unit is shown as `/ dL`. The others are in `µg/m³`. I assume it should say `µg/dL`. But I wonder why it's not also just `µg/m³`

MallocArray, 2024-01-28:
I'm not setting the units in the config, it is coming directly from ESPHome, so would need to have it changed in the ESPHome code on their issue page

https://esphome.io/components/sensor/pmsx003.html#configuration-variables

MallocArray, 2024-01-28:
After re-reading that page, I think it is correct.  It is the number of particles/dl instead of the other sensors that are in ug/m3

kroeckx, 2024-01-28:
Reading that, 0.3 doesn't even seem to be supported by the PMS5003.

MallocArray, 2024-01-28:
It isn't supported with the std parameters, but it is certainly supported and returns values using this:

pm_0_3um (Optional): Use the number of particles with diameter beyond 0.3um in 0.1L of air.

kroeckx, 2024-01-28:
So the datasheet seems to say it outputs:
- PM1.0 in µg/m³ (std particle)
- PM2.5 in µg/m³ (std particle)
- PM10  in µg/m³ (std particle)
- PM1.0 in µg/m³ (atmospheric environment)
- PM2.5 in µg/m³ (atmospheric environment)
- total(?) in µg/m³ (atmospheric environment)
- particles > 0.3 in particles/dL
- particles > 0.5 in particles/dL
- particles > 1.0 in particles/dL
- particles > 2.5 in particles/dL
- particles > 5 in particles/dL
- particles > 10 in particles/dL

kroeckx, 2024-01-28:
So I guess I would like to monitor all except the std ones, and it currently has pm_1_0, pm_2_5, pm_10_0, pm_0_3um.

MallocArray, 2024-01-29:
I don't think I'll add all of those by default, but users are able to configure their device however they want, so you can add as many or few of these as you desire.

Using that link, you can see what all of the configuration variables are.  

You could copy the contents of the starting package file in this repo and paste it into the bottom of your current .yaml file for your device.  
Then remove the line `  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003.yaml` 
Then make any modifications to the pasted lines, adding or removing sensors as you desire and then save and install.  You'll still get all of the other packages from my repo, but you get full control of the PMS5003 sensors

kroeckx, 2024-01-30:
At least calling pm_0_3um PM0.3 is confusing, it more closely matches PM10. That is, the PM1.0, PM2.5 and PM10 are about particles < the 1.0, 2.5 and 10 um size, while pm_0_3um is about particles > 0.3, which is also the smallest it can see.

MallocArray, 2024-01-30:
The name "PM 0.3" in my config is partially based on the fact that the parameter in ESPHome is called pm_0_3um which I can't control, and partially based on the AirGradient Dashboard calling it "PM 0.3 Count"

[image]

closed 2024-03-13 by MallocArray