MallocArray/airgradient_esphome #27: Add sleep mode

domoritz, 2024-01-29:
First off, awesome configuration. It makes my air gradient so much more useful since it now seamlessly integrates into my home assistant setup.

I read that the senseair sensors have a limited lifetime and can be set to update less often. Also, the LED and OLED display on the air gradient probably don't need to be on at night. It would be awesome to have a combined "sleep mode" that reduces brightness (similar to https://github.com/airgradienthq/arduino/pull/40 but manual via a switch in esphome) and update intervals.

domoritz, 2024-01-29:
You're right, I misspoke. I meant have a sleep mode for the Plantower PMS5003 sensor. I like your idea not to have a general sleep mode as it can be handled in home assistant instead. So I rewrote the issue to ask for a sleep mode/extended life for the Plantower sensor that can be toggled without recompiling.

MallocArray, 2024-01-29:
It will be taken into consideration.

Making it a substitution variable that you could define in your ESPHome yaml at install time is one option
Another option is to add a slider bar or a field you could type into that could fully control it, but adds new complexity, and currently we are nearly out of memory if adding the ESP32_Improv feature to allow for bluetooth discovery, so being a bit cautious about adding all of the features

I also just started this conversation on the AirGradient forums to see if we can get some testing about impacts to adjusting the interval and if it will affect accuracy overall
https://forum.airgradient.com/t/accuracy-comparison-of-pms5003-with-large-interval-values/1615

pyrodex, 2024-02-06:
Can you also do a pm_2.5_2 extended life yaml? I could copy and paste it myself but it would be nice to have yours referenced for folks who want to be native.

MallocArray, 2024-04-03:
> Can you also do a pm_2.5_2 extended life yaml? I could copy and paste it myself but it would be nice to have yours referenced for folks who want to be native.

I just added a package for an extended life version of the second PMS5003T sensor.  It is also set to 2 minutes.
sensor_pms5003t_2_extended_life.yaml

In recent forum discussions, the highest risk of using this with the outdoor model is that the fan will spin down after 30 seconds and could provide an opportunity for bugs to crawl inside.  Just be aware.

closed 2024-05-19 by MallocArray

domoritz, 2024-05-19:
Can you set the pm update interval at runtime now? I looked at the code and only saw how it can be changed at compile time.

MallocArray, 2024-05-19:
I'm pretty sure this is not possible, as the sensor needs to know the value at compile time.

If you find anything from ESPHome that indicates different, we can look at modifications to enable that, but I would need links to examples or documentation.