MallocArray/airgradient_esphome #30: O-1PPT Firmware still getting 400 bad request errors on upload to Airgradiant

jayofdoom, 2024-02-04:
Hey,

Just wanted to let you know, the 400 errors continue, it doesn't appear to be reporting to the airgradient API successfully. I also will note,  my machine has not emitted an AQI category (https://github.com/ajfriesen/ESPHome-AirGradient/blob/main/air-gradient-open-air.yaml#L189) in over a week. I suspect these may be related, somehow, but I've seen nothing in the logs to point to a problem.

Happy to help debug in any way I can.

Thanks,
Jay

jayofdoom, 2024-02-04:
`[01:50:13][W][http_request:093]: HTTP Request failed; URL: http://hw.airgradient.com/sensors/airgradient:84fce60bee00/measures; Code: 400; Duration: 1185 ms`

jayofdoom, 2024-02-04:
I got my tabs mixed up, somehow thought I was looking at your code there. I may add support for this :).

Sorry for the noise!

jayofdoom, 2024-02-04:
Is my proof of concept for supporting this if you want. I'm not PRing it because:
1) 

```
text_sensor:
    - platform: template
      # Depends on another sensor providing an ID of pm_2_5 such as a pms5003
      name: "PM 2.5 AQI Category"
      id: pm_2_5_aqi_category
      update_interval: 5 min
      icon: "mdi:air-filter"
      # AQI Categories are documented at: https://www.airnow.gov/aqi/aqi-basics/
      lambda: |-
        if (id(pm_2_5_aqi).state <= 50.0) {
          return (std::string) "Good";
        } else if (id(pm_2_5_aqi).state <= 100.0) {
          return (std::string) "Moderate";
        } else if (id(pm_2_5_aqi).state <= 150.0) {
          return (std::string) "Unhealthy for Sensitive Groups";
        } else if (id(pm_2_5_aqi).state <= 200.0) {
          return (std::string) "Unhealthy";
        } else if (id(pm_2_5_aqi).state <= 300.0) {
          return (std::string) "Very Unhealthy";
        } else {
          return (std::string) "Hazardous";
        }
```

MallocArray, 2024-02-06:
Thank you.  I'll take a look when I have some time to focus on it and see about adding it as another Package.

jayofdoom, 2024-02-06:
I'll note I found a bug in that code, I have to also check for NaN and return "Unknown" in that case (or else on startup you get a couple of Hazardous readings reported before it has data).

MallocArray, 2024-05-06:
This may help with the NaN status you mentioned:

```yaml
            - if:
                condition:
                  lambda: 'return (isnan(id(voc).state));'  # No data from sensor yet
                then:
```

MallocArray, 2024-05-19:
A new package file has been added today that calculates Nowcast and AQI including the Category.  You may check that out.
[sensor_nowcast_aqi.yaml](https://github.com/MallocArray/airgradient_esphome/blob/main/packages/sensor_nowcast_aqi.yaml)

If there is further discussion about that specific topic, please make a new Issue to continue there.

closed 2024-05-19 by MallocArray