MallocArray/airgradient_esphome #31: question: relationship to airgradienthq/arduino

domoritz, 2024-02-06:
What's the relationship of this setup with esphome to https://github.com/airgradienthq/arduino? Does it benefit from the improvement of the most recent [3.0 release](https://github.com/airgradienthq/arduino/releases/tag/3.0.0)?

MallocArray, 2024-02-06:
Good question.  This repo was originally created to be a close equivalent to the Arduino repo 2.x release in ESPHome, plus some quality of life improvements such as buttons to initiate a CO2 calibration and enable/disable automatic calibration.

I haven't reviewed the 3.x release yet.  I know there is some new functionality to allow settings in the AirGradient Dashboard to configure how the hardware behaves.  I'm not sure if that is possible in ESPHome, or if people would even want it, since ESPHome/HomeAssistant is also the main control dashboard that is completely in the end user's control.

So I'll review it at some point to see if some of the functionality could/should be brought into this repo.

domoritz, 2024-02-06:
Great. I don't need the configuration since ESPHome already does that but any kind of calibration improvements would be nice to have.

MallocArray, 2024-02-06:
For the Outdoor model, this repo does have the Temp/Humidity correction algorithm applied.

When I review the updated code, if there are any other compensations applied, I'll incorporate those.

closed 2024-02-09 by MallocArray