MallocArray/airgradient_esphome #32: Open Air 1-PPT isn't working

pyrodex, 2024-02-06:
I tried your configuration today and noticed the 1-PPT isn't working. 

I believe you have a GPIO mismatch as none of the sensors were reading at all and I reverted back to previous ESPHome configurations used from [air-gradient-open-air.yaml](https://github.com/ajfriesen/ESPHome-AirGradient/blob/main/air-gradient-open-air.yaml).

Thanks!

MallocArray, 2024-02-06:
If you would be willing to try again, after flashing, make sure to remove all power for 5-10 seconds and then reconnect.  Sometimes a reboot isn't enough to clear things that cause sensors to not report.

I don't have a 1-PPT myself, but I did unplug my CO2 sensor in 1-PST and connect a second PMS5003 sensor and all sensors were reporting.

I checked my pin config vs the one you linked and the pin numbers appear to match, so all I can think of is that maybe it needed a full power cycle after the flash.

pyrodex, 2024-02-06:
That seems to fix it, thanks!

closed 2024-02-09 by MallocArray