MallocArray/airgradient_esphome #33: Connection refused

kwdavidson, 2024-02-09:
I've been trying to get my AirGradient One to send data to the online dashboard, but the ESPHome log keeps showing the following when trying to connect:

`HTTP Request failed; URL: http://hw.airgradient.com/sensors/airgradient:<mac_address>/measures; Error: connection refused; Duration: 0 ms`

Does the API need you to use the token in the request?

MallocArray, 2024-02-09:
Connection Refused is a new one to me.  Usually it has an error code number.

Did you setup the Dashboard with the serial number of your device?
https://www.airgradient.com/documentation/diy-connect-to-dashboard/

You have to do that first so it will accept data from your device. You'll use the same mac_address that is in the URL

kwdavidson, 2024-02-09:
I assume the "serial number" is the MAC address. I've set this thing up three times on the Dashboard. It shows the correct commissioning date and a last seen date and time of when I entered it into the Dashboard. Then there are no updates and I get the connection refused messages. Should I bring this up with the AirGradient guys? It's working fine with Home Assistant, which displays the proper MAC address.

MallocArray, 2024-02-09:
Make sure it is all lower case and no special characters like :
If it still doesn't work, does it ever ask for a model number? If so, for the ONE it should be I-9PSL

Otherwise, I would e-mail support @airgradient.com to get assistance from them, as I don't believe it is ESPHome related

kwdavidson, 2024-02-09:
Yup. All lower case, no special characters, and model I-9PSL. I'll poke them. Thanks.

MallocArray, 2024-02-09:
Pretty sure it would be a different error, but do make sure you have DNS setup in your ESPHome config if you are using a static IP, as I forgot that at one point.  If you are using DHCP, then it shouldn't be an issue.

jhorn70, 2024-02-13:
I am having a similar issue, turning on update of the AG dashboard throws mine into a boot loop and I have to reflash.

MallocArray, 2024-03-13:
Are these issues still present?

kwdavidson, 2024-03-14:
I finally gave up after trying a bunch of things. I'd rather keep the data local anyway.

closed 2024-03-14 by MallocArray