MallocArray/airgradient_esphome #34: Thoughts about LED gauge?

joneshf, 2024-02-11:
Is there any interest in using the LEDs as a gauge? Right now, it seems that it uses all 11 LEDs at once, and they all have the same color (based on CO<sub>2</sub>). But using something like [light.addressable_set](https://esphome.io/components/light/#light-addressable-set-action) and [light.pulse_effect](https://esphome.io/components/light/#pulse-effect) it could also be something like:
| CO<sub>2</sub> | LEDs |
| --- | --- |
| < 800 ppm | 🟢🟢 |
| >= 800 ppm && < 1000 ppm | 🟢🟢🟢🟢 |
| >= 1000 ppm && < 1200 ppm | 🟡🟡🟡🟡🟡🟡 |
| >= 1200 ppm && < 1400 ppm | 🟡🟡🟡🟡🟡🟡🟡🟡 |
| >= 1400 ppm && < 1600 ppm | 🔴🔴🔴🔴🔴🔴🔴🔴🔴🔴 |
| > 1600 ppm | 🔴🔴🔴🔴🔴🔴🔴🔴🔴🔴 (slowly pulsing) |

Or leave the pulsing off, if that seems like too much. In any case, what do you think about something like this?

MallocArray, 2024-02-11:
I like the idea.  Would have to see how that feels in real life, if having only 2 green seems like it isn't being utilized.

Do you want to mock up the ESPHome config to enable it and we can see how it feels?  

I would probably match the levels to what is currently defined and the color, but adjust how many LEDs are used.  AirGradient released a new version of their Arduino code that I want to see if they changed the scales at all.

I also have plans to make an option to have the LED reflect the PM2.5 levels instead of CO2 so users have options.  I toyed around with having the bar split, so left half was CO2 and right half was PM2.5 but think that may end up confusing.

joneshf, 2024-02-11:
Okay cool. I threw something together yesterday without the pulsing (because I was lazy). Lemme change back the colors/thresholds and I'll toss it up.

Re: splitting. There's 11 LEDs. Maybe having each side be 5 LEDs with an unused one in the middle might be enough of a visual cue? Something like:
🟢🟢🟢🟢🟢⚪🟢🟢🟢🟢🟢

I actually kind of like the idea of showing two (or more) things. Might noodle around with that as well. Maybe slap a (removable) label under each section or something.

MallocArray, 2024-02-23:
Did you have some example code to share yet?

joneshf, 2024-03-01:
Oh also, all of the LEDs are on when first powering on the device. After the CO<sub>2</sub> is reporting and the interval has passed, the LEDs to change to this gauge. I'm not sure which thing would need to change to make it do something different.

scynthero, 2024-03-23:
I just bought AirGradient ONE and the default firmware has something similar if not straight the same as joneshf idea. It looks really nice.

DamonLane, 2024-04-07:
Seeing the row of LEDs, I assumed the acted like a gauge, and as scynthero mentioned, the default firmware does something like this. I didn't keep it on long enough to see what it was doing.

As a data analyst, I feel like this has to show at least two metrics, but we don't want to over-complicate it. It would seem intuitive to me if the gauge was the level of CO2 and the color was the PM. A discussion like this is good because what's intuitive is probably different for different people.

ossgeek314, 2024-04-29:
As @scynthero has mentioned, the default AirGradient ONE has the ability to indicates the PM2.5 levels with the LED strip using individual addressed lights and colors.  I recently switched to esphome from the default and really missed the quick indication for PM2.5 (my primary concern).  To that end, I threw together a yaml to recreate what is in the default AirGradient ONE firmware for pm2.5.  I'm no YAML expert and certainly very new to ESPHome so any efficiency suggestions or modifications would be greatly appreciated!

This could easily be recreated for CO2.  I was actually even thinking of using the first 10 LED lights for PM2.5 and then the last LED is just a color coded CO2 indicator.

Another option for further consideration would be creating a toggle (like the displays) to choose between CO2 and PM2.5 or having it split down the middle with CO2 on one side and PM2.5 on another.

At any rate, attached is the yaml I'm currently testing out for this purpose to just mimic the AirGradient ONE default firmware (I commented out the CO2 yaml from this repo for now)
[led_pm25.yaml.txt](https://github.com/MallocArray/airgradient_esphome/files/15149463/led_pm25.yaml.txt)

ossgeek314, 2024-04-30:
I decided to add the CO2 indicator to the last LED in the chain and utilize the first 10 as PM2.5 for my current setup.  One thing that I noticed was `led_brightness` became completely useless (for whatever reason `color_brightness` doesn't actually change the brightness for the `addressable_set` idk if this is a bug in esphome or a limitation of this particular LED strip).  Because of that I added a `led_brightness` trigger to the number template so that at least the brightness would be responsive.  It does light the entire led_strip when it changes brightness but then resets itself to the guage after the interval has passed.  Anyway, I'm attaching what I did hoping maybe there is a better way.
[image]

[led_co2.yaml.txt](https://github.com/MallocArray/airgradient_esphome/files/15163624/led_co2.yaml.txt)
[led_pm25.yaml.txt](https://github.com/MallocArray/airgradient_esphome/files/15163625/led_pm25.yaml.txt)
[led.yaml.txt](https://github.com/MallocArray/airgradient_esphome/files/15163626/led.yaml.txt)

joneshf, 2024-04-30:
> for whatever reason `color_brightness` doesn't actually change the brightness for the `addressable_set` idk if this is a bug in esphome or a limitation of this particular LED strip

I'm using `color_brightness` with `light.addressable_set`, and it changes the brightness of the LEDs. So it does seem to work on this hardware. You can try looking at my setup in https://github.com/MallocArray/airgradient_esphome/issues/34#issuecomment-1972933598 and seeing if anything jumps out at you. That said, the documentation for ESPHome leaves a lot to be desired, and debugging why something happened (or didn't happen) is pretty unintuitive. it's definitely confusing to figure out!

At the moment, I'm a little too lazy to download the files to check what's going on 😅 , but I could take a look if you wanted to paste the YAML in a code block. I'll probably be less lazy later today/this week and can look at the files, if you haven't figured it out before then.

joneshf, 2024-04-30:
I got less lazy…I don't see `color_brightness` anywhere in the files you attached, so it's a little hard to figure out what wasn't working.

I see you've got stuff like:
```YAML
light.addressable_set:
  id: led_strip
  range_from: 0
  range_to: 0
  red: 0%
  green: 100%
  blue: 0%
```
Does slapping in a `color_brightness` like this work?:
```YAML
light.addressable_set:
  color_brightness: !lambda 'return id(led_brightness).state / 100.0;'
  id: led_strip
  range_from: 0
  range_to: 0
  red: 0%
  green: 100%
  blue: 0%
```

Or was that what you already tried?

ossgeek314, 2024-04-30:
@joneshf ,

I removed the `color_brightness` from the yaml because it seemed to not be working (and I wanted to reduce leftover testing cruft).

When I was testing it I used the following:

```
light.addressable_set:
  id: led_strip
  range_from: 0
  range_to: 0
  color_brightness: !lambda 'return id(led_brightness).state / 100.0;'
  red: 0%
  green: 100%
  blue: 0%
```
I'm wondering now if I just didn't wait 60 seconds for it to change, expecting it to be instant response.

ossgeek314, 2024-05-01:
I just tried with the `color_brightness` attribute but was patient and allowed the minute interval to pass and it worked as expected.

Would be cool to find a way to make the response more immediate.

ossgeek314, 2024-05-02:
I may have spoken too soon.  I setup an automation in HomeAssistant to decrease the brightness at night (25%) this seemed to completely turn off the light (but that might just be the light not able to shine through the plastic).  In the morning it automatically set the brightness to 80% but the lights have yet to come on even after a few hours.

I was able to bring the LEDs back to life by toggling the LED switch off and then on.

MallocArray, 2024-05-03:
@DamonLane could you share your automation for HA that is doing the dimming and resuming?

I've been meaning to get back to setting up this automation as well as making the LED strip be optionally PM2.5 or CO2, as well as a combined meter as discussed here.

I don't really care for the default AirGradient behavior as I find it very unintuitive and underutilizing the strip.  I've only ever seen the 4 far right LEDs used.

I'm going to take a look at what @ossgeek314 has posted and see if I want to put my spin on it and put a PR for you all to review

DamonLane, 2024-05-04:
I'm new to Home Assistant so this may not be the best way, but it works. I set it up in the GUI, but the yaml is probably easier to share than screenshots.

This turns my four indoor monitors' LEDs on at 7 am. Although I only set the brightness (to zero) at night, I have to switch them on, in addition to increasing their brightness for them to come back on. 

```
alias: AG LED strips on in the morning
description: ""
trigger:
  - platform: time
    at: "07:00:00"
condition: []
action:
  - service: light.turn_on
    metadata: {}
    data: {}
    target:
      entity_id:
        - light.ag_71_2_led_strip
        - light.ag_73_1_led_strip
        - light.ag_73_2_led_strip
        - light.ag_71_1_led_strip
  - service: number.set_value
    metadata: {}
    data:
      value: "40"
    target:
      entity_id:
        - number.ag_71_2_led_brightness
        - number.ag_73_1_led_brightness
        - number.ag_73_2_led_brightness
        - number.ag_71_1_led_brightness
mode: single
```

And this dims their LED strips at sunset

```
alias: Dim AG LED at sunset
description: testing 71-2
trigger:
  - platform: sun
    event: sunset
    offset: 0
condition: []
action:
  - service: light.turn_off
    target:
      entity_id:
        - light.ag_71_2_led_strip
        - light.ag_73_1_led_strip
        - light.ag_73_2_led_strip
        - light.ag_71_1_led_strip
    data: {}
mode: single
```

MallocArray, 2024-05-04:
[led_combo.yaml.txt](https://github.com/MallocArray/airgradient_esphome/files/15210234/led_combo.yaml.txt)

Here is my first attempt at a combo LED bar, based on the information shared here.  See what you think.
Increased update speed to 5 seconds
# Left 5 LEDs indicate CO2
# Middle 5 LEDs indicate PM2
# Far right LED indicates TVOC
If no data from sensor, turns off the LED until good data is received (Keeps it from going full brightness when no valid data)

My idea for this is it matches the layout of the display so you can reference what each section means, but with the LED bar, you can easily tell across the room.

I also made a config for the full bar representing PM2.5 like the current CO2 one does.
If there is interest in following exactly what AirGradient firmware does, that can be created as well.

MallocArray, 2024-05-06:
I have a new branch started with modifications to the LED behavior.  You can try out the latest by adding the branch to the end of the package path.  Remove/comment out the led_co2 line and add this to try out the latest, or pull the config from the branch:

`  led_combo: github://MallocArray/airgradient_esphome/packages/led_combo.yaml@led_strip_improvements`

In this one, instead of having 5 steps where the LEDs are either Green, Yellow, Orange, Red, or Purple, it will now blend/fade between the colors, so if the sensor is between the set points, the colors will be a between color.

The `substitutions` at the top of the file controls at what sensor values it should be at each full color, so you can tweak by adding the relevant line to a substitution section in your main yaml, no need to copy values you don't want to change.

In this same branch I made the same changes to the files for the full bar displaying CO2 or PM 2.5.

MallocArray, 2024-05-07:
Pushed another update. Now there is an "LED Fade %" value you can set in HomeAssistant, so on the 5 LEDs for CO2 or PM2.5, it can fade out by that percentage.  So the Center LED is the set brightness, the LEDs on each side are 20% dimmer and the LEDs on the sides of those 2 are 20% dimmer again by default.

As you lower the overall LED bar brightness, the fade amount needs to be decreased or you can't see it at all when LED Brightness is 50%.

Or you can set Fade to 0 and it will be all full brightness.

MallocArray, 2024-05-13:
My take has been included via 319c24520c371954c39f97f39b1f9c4bdcdf8fee for the full bar for CO2 and PM2.5, as well as the combo bar shown in the screenshot.

I will still work on full instructions for setting up an automation in HA for disabling the LED and display.  I think it will be easiest to do it in HA as an Automation rather than try to have it all internal to ESPHome.

closed 2024-05-13 by MallocArray