MallocArray/airgradient_esphome #40: Airgradient DIY PRO - Unable to connect from ESPhome if using display_sh1106_multi_page package

Vertikar, 2024-03-12:
## Describe the bug
When multi page is used, unable to connect from ESPHome or Home Assistant.
Device seems to be functioning ok, and updates values on the screen as expected. Does only seem to switch between two screens though, looks like the boot screen, and the original Airgradient display,

## To Reproduce
switch from display_sh1106_single_page.yaml package to display_sh1106_multi_page.yaml

## Logs
This warning at install/compile didn't seem related to me, but if I switch back to the single package it compiles without warnings and runs fine.
```
|-- Sensirion Gas Index Algorithm @ 3.2.1+sha.dbf2982
|-- ESP8266HTTPClient @ 1.2
Compiling .pioenvs/airgradient-pro/src/main.cpp.o
/config/.esphome/packages/76b4402f/packages/display_sh1106_multi_page.yaml: In lambda function:
/config/.esphome/packages/76b4402f/packages/display_sh1106_multi_page.yaml:132:37: warning: zero-length gnu_printf format string [-Wformat-zero-length]
  132 |           it.printf(0, 0, id(open_sans_14), "");
      |                                     ^~
Linking .pioenvs/airgradient-pro/firmware.elf
RAM:   [======    ]  61.8% (used 50608 bytes from 81920 bytes)
Flash: [=======   ]  69.0% (used 720189 bytes from 1044464 bytes)
Building .pioenvs/airgradient-pro/firmware.bin
esp8266_copy_factory_bin([".pioenvs/airgradient-pro/firmware.bin"], [".pioenvs/airgradient-pro/firmware.elf"])
========================= [SUCCESS] Took 46.95 seconds =========================
```

This is the error when ESPHome attempts to connect, and see a similar error in Home Assistant
```
INFO Upload took 18.96 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.0.152 using esphome API
WARNING Can't connect to ESPHome API for airgradient-pro @ 192.168.0.152: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.0.152', port=6053))] (TimeoutAPIError)
INFO Trying to connect to airgradient-pro @ 192.168.0.152 in the background
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.013s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: airgradient-pro @ 192.168.0.152: EOF received
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.013s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: airgradient-pro @ 192.168.0.152: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.2.2).
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.021s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: airgradient-pro @ 192.168.0.152: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.2.2).
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.011s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: airgradient-pro @ 192.168.0.152: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.2.2).
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.005s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: airgradient-pro @ 192.168.0.152: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.2.2).
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.006s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: airgradient-pro @ 192.168.0.152: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.2.2).
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.008s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: airgradient-pro @ 192.168.0.152: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.2.2).
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.022s
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.046s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: [Errno 104] Connection reset by peer
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.032s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: airgradient-pro @ 192.168.0.152: Connection lost
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.013s
WARNING airgradient-pro @ 192.168.0.152: Connection error occurred: airgradient-pro @ 192.168.0.152: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.2.2).
INFO Successfully connected to airgradient-pro @ 192.168.0.152 in 0.011s

```

MallocArray, 2024-03-13:
I haven't seen that compile error myself, but I'll watch it again closer.

By default, even with the multi-page package used, it only defaults the 2 pages you mentioned as being enabled.  In HomeAssistant, you can go to Devices and find this device.  Under the Configuration section, you can flip the switch to enable the different pages and it will cycle through any you have turned on

As to why you can't get it to connect to ESPHome or HA when using that package, I'm not able to reproduce that either, but wonder if it is related to the compile error.  Did you make any other changes to the config that may be consuming more memory, such as enabling the ble_improve or adding any other sensors?

[image]

Vertikar, 2024-03-13:
The only difference is that I have it syncing time with Home Assistant, and I've got an older board so had to switch to the SHT30 sensor.

I did try it with the default - SHT40 replacing the 30, and was still able to reproduce it, but haven't tried disabling the time sync.
will have a try with that and report back.

This is my current config:

```
# AirGradient Pro V3.3 - V4.2
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  name: "airgradient-pro"
  friendly_name: "airgradient-pro"
  config_version: 2.0.1
  name_add_mac_suffix: "false"  # Must have quotes around value

# Enable Home Assistant API
api:
  encryption:
    key: <removed>
ota:
  password: <removed>
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Airgradient-Pro Fallback Hotspot"
    password: <removed>

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-pro.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_d1_mini_board.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  temp_humidity: github://MallocArray/airgradient_esphome/packages/sensor_sht30.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  display: github://MallocArray/airgradient_esphome/packages/display_sh1106_single_page.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_d1_mini.yaml
  config_button: github://MallocArray/airgradient_esphome/packages/config_button.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
  safe_mode: github://MallocArray/airgradient_esphome/packages/switch_safe_mode.yaml

binary_sensor:
  - id: !extend config_button
    pin:
      number: D7
```

> By default, even with the multi-page package used, it only defaults the 2 pages you mentioned as being enabled. In HomeAssistant, you can go to Devices and find this device. Under the Configuration [...]

Vertikar, 2024-03-13:
Same thing even with time sync disabled.

```
INFO ESPHome 2024.2.2
INFO Reading configuration /config/airgradient-pro.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing airgradient-pro
(board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------

HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- Wire @ 1.0
|-- ArduinoJson @ 6.18.5
|-- Sensirion Gas Index Algorithm @ 3.2.1+sha.dbf2982
|-- ESP8266HTTPClient @ 1.2
Compiling .pioenvs/airgradient-pro/src/main.cpp.o
/config/.esphome/packages/76b4402f/packages/display_sh1106_multi_page.yaml: In lambda function:
/config/.esphome/packages/76b4402f/packages/display_sh1106_multi_page.yaml:132:37: warning: zero-length gnu_printf format string [-Wformat-zero-length]
 132 | it.printf(0, 0, id(open_sans_14), "");
 | ^~
Linking .pioenvs/airgradient-pro/firmware.elf
RAM: [====== ] 61.4% (used 50272 bytes from 81920 bytes)
Flash: [======= ] 68.2% (used 712397 bytes from 1044464 bytes)
Building .pioenvs/airgradient-pro/firmware.bin
esp8266_copy_factory_bin([".pioenvs/airgradient-pro/firmware.bin"], [".pioenvs/airgradient-pro/firmware.elf"])
========================= [SUCCESS] Took 44.96 seconds =========================

INFO Successfully compiled program.
INFO Connecting to [192.168.0.152](http://192.168.0.152/)
INFO Uploading /config/.esphome/build/airgradient-pro/.pioenvs/airgradient-pro/firmware.bin (716544 bytes)
INFO Compressed to 491890 bytes
Uploading: [============================================================] 100% Done...

INFO Upload took 19.82 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from [192.168.0.152](http://192.168.0.152/) using esphome API
WARNING Can't connect to ESPHome API for airgradient-pro @ [192.168.0.152](http://192.168.0.152/): Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='[192.168.0.152](http://192.168.0.152/)', port=6053))] (TimeoutAPIError)
INFO Trying to connect to airgradient-pro @ [192.168.0.152](http://192.168.0.152/) in the background
INFO Successfully connected to airgradient-pro @ [192.168.0.152](http://192.168.0.152/) in 0.007s
WARNING airgradient-pro @ [192.168.0.152](http://192.168.0.152/): Connection error occurred: [Errno 104] Connection reset by peer
INFO Successfully connected to airgradient-pro @ [192.168.0.152](http://192.168.0.152/) in 0.012s
WARNING airgradient-pro @ [192.168.0.152](http://192.168.0.152/): Connection error occurred: [Errno 104] Connection reset by peer
INFO Successfully connected to airgradient-pro @ [192.168.0.152](http://192.168.0.152/) in 0.016s
WARNING airgradient-pro @ [192.168.0.152](http://192.168.0.152/): Connection error occurred: [Errno 104] Connection reset by peer
INFO Successfully connected to airgradient-pro @ [192.168.0.152](http://192.168.0.152/) in 0.006s
WARNING airgradient-pro @ [192.168.0.152](http://192.168.0.152/): Connection error occurred: [Errno 104] Connection reset by peer
INFO Successfully connected to airgradient-pro @ [192.168.0.152](http://192.168.0.152/) in 0.007s
WARNING airgradient-pro @ [192.168.0.152](http://192.168.0.152/): Connection error occurred: airgradient-pro @ [192.168.0.152](http://192.168.0.152/): EOF received
INFO Successfully connected to airgradient-pro @ [192.168.0.152](http://192.168.0.152/) in 0.015s
WARNING airgradient-pro @ [192.168.0.152](http://192.168.0.152/): Connection error occurred: airgradient-pro @ [192.168.0.152](http://192.168.0.152/): The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.2.2).
INFO Successfully connected to airgradient-pro @ [192.168.0.152](http://192.168.0.152/) in 0.005s
```

MallocArray, 2024-03-13:
What versions of HomeAssistant are you running?  If it is older than a few months, would you be up for updating it?

On the compile error,  I see it is saying there is a zero length field on line 132 that I'll have to take a closer look at to see what it has, but I'm away from my primary desk

it.printf(0, 0, id(open_sans_14), "");

MallocArray, 2024-03-14:
I was able to reproduce the error about the zero-length string, but it only appears on the ESP8266 based board, and not on the ESP32-C3 device.
I'll be pushing up a new revision soon.  You may not see it until after 24 hours has elapsed, due to ESPHome caching packages from external sources.

We can see if it has anything to do with your lack of connectivity.  I don't have any API or OTA encryption key set on my unit, so that is one difference that I would have to do a bit of work to get setup to match your config, so not tonight

MallocArray, 2024-03-14:
Scratch that, I am able to reproduce the issue where I can still view logs in ESPHome and the display is showing output, but it is not connecting to HomeAssistant.
I'll keep looking at it.

MallocArray, 2024-03-14:
You are current enough.  Pretty sure the ESP8266 is running out of RAM with the extra fonts and/or pages defined.  Wasn't this bad of an issue with earlier releases, but I now see the same as what you are seeing.

Looking at the Heap Free memory, you can see when I flipped to multi-page around 8:00 pm yesterday and the amount of free memory went down significantly.

I'll work on reducing the number of different fonts to see if that help.  But honestly, I've liked the single page view so much, I haven't been using the multi-page lately.

[image]

Relevant link: https://community.home-assistant.io/t/esphome-connection-error-occurred-eof-received/603115/4

Vertikar, 2024-03-15:
> but I now see the same as what you are seeing.

Ah glad you can reproduce it.

> But honestly, I've liked the single page view so much, I haven't been using the multi-page lately.

I do really like it too, but I've found in the past I've got major burn in on OLEDs like this one when they've always been displaying roughly the same thing, so was thinking could avoid it with multi-page

Vertikar, 2024-04-04:
Great, that's sorted it out for me too.
Thanks for fixing that up!

closed 2024-04-04 by Vertikar