MallocArray/airgradient_esphome #44: Is it possible to have the LED strip change based on AQI instead of CO2? Or combo

bigrjsuto, 2024-03-14:
Was just thinking about this. 

is this possible?

If not, how difficult would it be to implement? I'm a novice when it comes to coding so I doubt it's something I'd be able to tackle if it's not very easy to do.

MallocArray, 2024-03-14:
It is possible. I plan on doing an option based on PM2.5 values which correlate to AQI, but I wasn't planning on using AQI directly.

I'm not sure if it should be a toggle switch in HomeAssistant to flip, or just a config file change to flip between them.  I'm considering the config file version so prevent the firmware from getting too overloaded

bigrjsuto, 2024-03-19:
Whatever is easiest!

Let me know when you do implement it.

Thanks for your great work.

closed 2024-03-19 by bigrjsuto