MallocArray/airgradient_esphome #45: Airgradient ONE: Unable to connect to AP on 2.4Ghz channel 13; works with default firmware

mgc8, 2024-03-15:
I recently received one of the Airgradient ONE kits and have been testing it with their dashboard which worked nicely, but then wanted to connect it to Home Assistant. To that end, I flashed the `airgradient-one.bin` file from here to the device, which worked nicely and it shows the sensor values on the OLED display.

However, when connecting to the `ag-one` hotspot to set up the WiFi, I am unable to select my AP -- which is on 2.4Ghz channel 13 (EU-based). This worked perfectly fine with the default firmware, so it's not a hardware limitation. Is there a configuration/region lock change that can be made?

MallocArray, 2024-03-16:
I'm not seeing any configuration options to enable EU channels.  You could try specifying your Wifi settings in the YAML file so you avoid the hotspot entirely. 
Something like:
```yaml
wifi:
  ssid: MySSID
  password: mypassword-here
```
https://esphome.io/components/wifi.html#connecting-to-multiple-networks

Since I'm not doing anything specifically to limit it to US channels, it might be best to ask on the ESPHome support options to get a wider audience:
https://community.home-assistant.io/latest
https://github.com/esphome/issues
https://discord.gg/xJHByb94

mgc8, 2024-03-16:
Thank you for the quick reply. It seems I need to dig deeper into the configuration options to find out where that limitation is in that case. I'll switch back to the default firmware in the meantime and investigate the alternatives.

MallocArray, 2024-06-29:
If anything, this appears to be an issue with ESPHome itself, so if continued effors are needed, please open an issue there

https://github.com/esphome/issues/issues

closed 2024-06-29 by MallocArray