MallocArray/airgradient_esphome #46: On-device AQI calculations

ex-nerd, 2024-04-12:
When I got my OpenAir, I added on-device AQI calculation to existing firmware I found. Since it resets if you power cycle, it's not as good as getting this inside of home assistant but so far I haven't found a plugin that actually does that (and I got side-tracked while working on the necessary database queries to do it myself).

The main difference with my calculations are that they are intended to more closely follow official AQI and NowCast calculations (specific differences are documented in [this commit](https://github.com/ex-nerd/ESPHome-AirGradient/commit/3c0e8db922cfb3de7950fb34aa9874739f56c95f)), which require 24 and 12 hours worth of data, respectively.

You can find the code here:

https://github.com/ex-nerd/ESPHome-AirGradient/blob/main/air-gradient-open-air.yaml

I haven't had time to poke through your code yet to figure out the best place to add it, but if someone else doesn't beat me to it, I will try to get a pull request put together to port this over. I also haven't done any work for a proper "indoor AQI" calculation, which I know includes a mix of info besides just particulate measurement.

MallocArray, 2024-04-15:
I wonder if doing the AQI inside of HomeAssistant as a sensor template would be better, as it keeps the complex configuration out of ESPHome and lets the data reside in HA's database instead of in-memory of the ESP device, which is pretty limited, especially the ESP8266 based devices.

Looks like your code is very similar to https://github.com/ajfriesen/ESPHome-AirGradient/blob/main/air-gradient-open-air.yaml
Did you contribute similar code there?

I haven't read all the way through your code, but I see you aren't restoring the values of some of the globals, so not sure how you are tracking across a reboot, but I am trying to be cautious about the memory usage, as I've bumped up against the limit on the D1 Mini several times just with a handful of fonts.

ex-nerd, 2024-04-15:
> Looks like your code is very similar to https://github.com/ajfriesen/ESPHome-AirGradient/blob/main/air-gradient-open-air.yaml Did you contribute similar code there?

Yes. The code in that repo is from my pull request and should be identical.

And no, it doesn't track across reboot. Based on my research at the time, the flash memory on most esp32 devices can't handle very many write cycles and would only last a few months months if anything was being written at the rate that measurements are taken. I never got around to looking into longer write cycles because at "every 6 hours (or so)" you're just getting into the realm of time it takes for the NowCast to regenerate on its own.

I agree it's something best tracked within Home Assistant, though I still haven't had time to figure that out (it gets a bit messy calculating averages when that database only stores values when they change). I'll get to it eventually (if someone else doesn't do it first) but for now, these on-device calculations work well enough.

MallocArray, 2024-05-14:
I just pushed a package with your local nowcast and aqi calculations.  I'm still waiting for enough time for it to run the calculations, but the sensors all show up.

You could try it out by adding this under the packages section
`  nowcast_aqi: github://MallocArray/airgradient_esphome/packages/sensor_nowcast_aqi.yaml@nowcast_aqi`

ex-nerd, 2024-05-14:
Thanks. I'll have to poke at this after work.

Over the weekend, I started looking into what it would take to get proper AQI/Nowcast calculations into home assistant, and there just doesn't seem to be a way to reliably do this with just the SQL queries or the statistics sensor (which can't look at discrete time ranges in the past, only "from now to past-time"). In the mean time, I created https://github.com/ex-nerd/home-assistant-aqi so I can eventually do the necessary expansion of "value hasn't changed" but there's a lot to learn for how to set up a plugin in general so it'll be awhile because that has anything useful.

ex-nerd, 2024-05-14:
Oh, and just in case there is ever any doubt … the original project that contains my AQI calculation code has an MIT license, which should be compatible with this project's GPL. But I also grant separate GPL license for any of my own code that's used for this project.

MallocArray, 2024-05-15:
I have sporadic luck with uptime in general, but not sure if it is ESPHome overall, or a specific component.
Mine has been up for 20 hours now without issue, but I have other devices that often reboot every 2 hours.

Sometime a full power cycle by removing the power cord can help.  Other times if I turn off the AirGradient Upload, or comment out the entire `airgradient_api` line it can help stabilize, but not always.

[image]

ex-nerd, 2024-05-15:
Well, yes, it would seem there's a problem with http upload. Looks like there's a workaround posted here: https://github.com/esphome/issues/issues/2853

ex-nerd, 2024-05-15:
Just for posterity, here is the display I use in Home Assistant. It needs `card_mod`, `mini-graph-card`, `stack-in-card` extras from HACS:

```yaml
type: custom:stack-in-card
mode: vertical
keep:
  background: true
  border_radius: false
  margin: false
cards:
  - type: markdown
    content: >
      {%- set category =

      states('sensor.airgradient_open_air_nowcast_category')

      -%}

      {% if category == "unknown" %} # NowCast: Calculating

      {% set mins =
      states('sensor.airgradient_open_air_nowcast_minutes_remaining')

      | int %} Available in {% if mins > 60 %}{{ mins // 60 }} hours {% endif

      %}{{ mins%60 }} minutes

      {% else %}

      # NowCast: {{ category }}

      {% endif %}
    card_mod:
      style: |
        ha-card {
          {% set aqi = states('sensor.airgradient_open_air_nowcast') -%}
          {%- set aqi = aqi | float if is_number(aqi) else -1 -%}
          {%- if aqi < 0 -%}
            background-color: #a0a0a0; color: #393939;
          {%- elif aqi < 50 -%}
            background-color: #00e400; color: #393939;
          {%- elif aqi < 100 -%}
            background-color: #ffff00; color: #004000;
          {%- elif aqi < 150 -%}
            background-color: #ff7e00; color: #404000;
          {%- elif aqi < 200 -%}
            background-color: #ff0000; color: #400000;
          {%- elif aqi < 300 -%}
              background-color: #8f3f97; color: #300030;
          {%- else -%}
            background-color: #7e0023; color: #200000;
          {% endif %}
          text-align: center;
        }
        h1 {
          line-height: 1px;
          margin: 0 !important;
          padding: 0 !important;
          border: 1px solid red !impotant;
        }
  - type: conditional
    conditions:
      - entity: sensor.airgradient_open_air_nowcast
        state_not: unknown
    card:
      type: gauge
      entity: sensor.airgradient_open_air_nowcast
      needle: true
      min: 0
      max: 500
[... 171 lines cut ...]
        min_bound_range: 10
        show:
          labels: false
          fill: fade
          name: true
          icon: false
          name_adaptive_color: true
      - type: custom:mini-graph-card
        entities:
          - entity: >-
              sensor.airgradient_open_air_pm_10_0_average
            name: PM10
            state_adaptive_color: true
            color: burlywood
        line_width: 2
        hours_to_show: 24
        min_bound_range: 10
        show:
          labels: false
          fill: fade
          name: true
          icon: false
          name_adaptive_color: true
columns: 1
```

MallocArray, 2024-05-15:
Looking again, I see the "PM 2.5 1h Average" has a siliding window moving average that assumes the update_interval of the PMS5003 sensor is 3 minutes as you had it.  I'm currently using the ESPHome default which is whenever a value comes in, 1 second with this hardware.
There is an alternative file that has a 2 min interval configured.  There is some concern that allowing the fan to shut down outdoors could let insects get inside or that accuracy on the first reading after spinup may not be fully accurate, so I don't want to hard code the default to 3 min.

Not sure what is best to do with the 1h average.  Leaving it as you had it wouldn't be a true 1 hr average if readings are coming in every second, but I don't think you can use a substitution for the window_size, or not with a calculation.

Thoughts?

ex-nerd, 2024-05-15:
> There is an alternative file that has a 2 min interval configured. There is some concern that allowing the fan to shut down outdoors could let insects get inside or that accuracy on the first reading after spinup may not be fully accurate, so I don't want to hard code the default to 3 min.

I wondered why the default was set to 1s when everything in the forums when I bought mine said to use the delayed values (especially with something like AQI that's intended to only change once/day, or nowCast once/hour). 

I think I added the [`pm_update_interval` substitution](https://github.com/ex-nerd/ESPHome-AirGradient/blob/main/air-gradient-open-air.yaml) to my original updates for this to account for that. Would it work to just add something like that to each of the two PMS5003 files? If not, maybe a global/variable would work.

ex-nerd, 2024-05-16:
> Not sure what is best to do with the 1h average. Leaving it as you had it wouldn't be a true 1 hr average if readings are coming in every second, but I don't think you can use a substitution for the window_size, or not with a calculation.

Looking at the code again, I wonder if the better option would be to replace the "copy" sensor with a template that runs on a schedule and just reads the current value of the sensor? Something like:

```yaml
  - platform: template
    id: pm_2_5_1h_avg
    name: "PM 2.5 1h Average"
    disabled_by_default: true
    device_class: pm25
    accuracy_decimals: 1
    filters:
      - sliding_window_moving_average:
          window_size: 60
          send_every: 60
          send_first_at: 60
    update_interval: 60s
    lambda: |
      return id(pm_2_5_1h_avg).state;
    on_value:
      lambda: |
        // Insert the current value
        float current = id(pm_2_5_1h_avg).state;
        if (!isnan(current)) {
          id(pm_2_5_hourly_avg).insert(id(pm_2_5_hourly_avg).begin(), current);
          // Truncate anything past the first 24
          if (id(pm_2_5_hourly_avg).size() > 24) {
            id(pm_2_5_hourly_avg).resize(24);
          }
        }
```

MallocArray, 2024-05-16:
My uptime is currently 5 hours but I have and AQI number and a status of Moderate.  Did you check your uptime to see if it rebooted recently?

I see what you are doing with the template sensor and I think that would work.

MallocArray, 2024-05-16:
Just pushed a new update with the modifications and setup a substitution for the update_interval on the extended_life configs.

ex-nerd, 2024-05-16:
> Did you check your uptime to see if it rebooted recently?

A reboot would/should reset the counter. Counter reached 0min right at 18h of uptime (total now of 27h) but the AQI values were never received. Looks more like something just didn't send the AQI number or category name. I'm running the "extended life" version of the PMS5003 code so I wonder if it's something in the math messing up the number of entries in the rolling window counter.

… though looking at the NowCast counter, it actually ended early (about 9h instead of 12h, probably due to the 2.5min vs 3min differences in your extended-life readings and the original ajfriesen ones). I'd expect the standard AQI calculation to also be faster. That also suggests there's something not quite right with the sliding window on the `copy` sensor config.

Either way, it looks like the once/hour calculation for those sensors may be too infrequent. I originally set those to avoid flooding Home Assistant with duplicate values, but I now know that HA basically ignores updates if the value doesn't change, so it's probably safe to increase the frequency [here](https://github.com/MallocArray/airgradient_esphome/blob/nowcast_aqi/packages/sensor_nowcast_aqi.yaml#L387) to something like 5min or 15min.

ex-nerd, 2024-05-16:
> Just pushed a new update with the modifications and setup a substitution for the update_interval on the extended_life configs.

Installed/running (even remembered to double check that the cache updated … apparently just wiping build files isn't actually enough by itself).

ex-nerd, 2024-05-17:
> I 100% think it's best to do this with a template sensor rather than try to shove it on the device TBH.

I've finally gotten around to working on a [plugin to do this](https://github.com/ex-nerd/home-assistant-aqi), but I have to get through the initial learning curve of "how to write a plugin" first (and find time among a dozen other hobbies).  Unfortunately, there just isn't a way to "expand" the HA readings using basic sensors and create accurate averages. Your "noisy" trick sounds interesting but does result in a huge number of readings that could eventually slow the whole database down and defeat the purpose of the original devs' "compressing" readings in the first place.

The on-device calculations already existed from my earlier work, and as of yesterday's update (esp. with the addition of the http-post crash workaround), everything seems stable and running how it did on the other firmware. It's annoying that there's a delay when the device power cycles, but also requires near zero setup from the end user to get the values into HA.

Makes for a nice dashboard, too (this is from the code I put into an earlier comment):
[image]

ex-nerd, 2024-05-17:
> Just pushed a new update with the modifications and setup a substitution for the update_interval on the extended_life configs.

This also seems to have fixed (at least for this one boot cycle) the issue where the AQI counter went to zero but never started calculating data. This basically confirms my suspicion that it had something to do with the frequency messing up some of the math in the lambda that was rotating values out of the (for lack of a better word) ring buffer of recent hourly readings.

ex-nerd, 2024-05-18:
> To your point about thrashing the DB -- you must add any such 'noisy' sensors to your configuration exclusions in the recorder as I stated above.

How does that work with AQI, then? By definition, AQI is a rolling average of the last 24 hourly averages (picking the worst of pm2.5 or pm10 with a minimum of at 18 hourly readings). It needs that history. NowCast is similar, though over 12 hours and is weighted toward more recent readings so you need to keep track of the hours when readings were taken. The formulas are standard, though I remember when I was researching this stuff that the weights used for NowCast seem to be interpreted differently by different companies (PurpleAir tends to weight "more recent" stronger than others).

If you're not storing the "jitter" date and averages in the HA database then it's not really any different than doing the calculations on the device. The benefit of something in HA is that if you lose power for a couple seconds/hours, you don't lose data and have to wait 12-18h for readings to show up again. 

Anyway I've been running the on-device calculations since I wrote them last year ([git says june of 2023](https://github.com/ex-nerd/ESPHome-AirGradient/commit/3c0e8db922cfb3de7950fb34aa9874739f56c95f)). Nothing is stored on the device so there isn't really any wear/tear. I actually looked into it when I originally wrote the calculations but sadly, the flash on the esp* processors can't handle the write cycles required to maintain state across reboots … even once/hour storage adds up and could degrade the memory after half a year or so (could do so much more if these devices had microSD cards). Even at the "extended life" 2.5min check cycle, I would expect the fans in the particle sensors to die long before anything else.

ex-nerd, 2024-05-18:
On the other hand, storing hourly averages "with jitter" would probably be fine for HA and simplify some of the other calculations needed. Might even be able to get away with template sensors instead of a full plugin.

MallocArray, 2024-05-18:
So where are we at this point? Is the file in the current branch sufficient, particularly for @ex-nerd since he created the original method?

I personally don't even know what Nowcast means, and I can easily pull AQI from local sources, so I'm not sure that I'll be using this myself, but if it looks like what you are expecting, we can merge it in.  Or we can keep tweaking

ex-nerd, 2024-05-19:
It's been stable/accurate for me since I flashed your updated version yesterday, so … seems good to go for me. Can always file a bug report/PR if something doesn't work right.

Conversation about AQI calculations with Home Assistant template sensors is probably best handled elsewhere like the Home Assistant forum. I'll probably start a thread there for info once you've merged the AQI branch.

---

As for what the two readings are, my understanding is:

Official AQI is calculated once per day off of based on hourly averages of no fewer than 18 hours of the previous day. If you don't have at least 18 full hours of data, AQI for that day is supposed to be invalid. It was designed to measure air quality over long durations (weeks/months), e.g. "healthy forest air" vs "smoggy city air" … All of the airgradient calculations I've seen (including mine) avoid this once/day calculation and do it based on rolling averages. In the case of my algorithm, it starts counting after 18 hours worth of data but includes up to 24h (since the data is lost on reset, there's no need to check if any individual hour in the middle was missing data, and remove it from the calculation)

Because AQI is so (intentionally) slow to update, someone came up with NowCast as a way to represent "current conditions" … especially (specifically?) for wildfire smoke. NowCast looks at the previous 12 hours worth of hourly averages and uses a weighted scoring system (recent hours weigh more and taper off for each hour). This provides a sort of "hourly air quality" with a smoothing effect to make it somewhat resilient to plumes of smoke (or of clean air during a wildfire) passing over the exact spot of the sensor. It's more of a "what's the air quality like this afternoon" rather than "what's the air quality right this instant that you can get directly from the pm2.5 reading." I've heard that some providers use different weights for the hours, e.g. rumors are that PurpleAir weights the most recent hour significantly heavier than others to give it more of a "real time" feel.

It's a pretty safe bet to assume that almost everywhere you see "AQI" being used to talk about air quality, it's actually NowCast.

Both calculations look at pm2.5 and pm10 and take the worst value for any given hour. As far as I've seen there are no official formulas that include readings for smaller particulate, nor for VOCs. Wikipedia has a good breakdown of the formulas for both[AQI](https://en.wikipedia.org/wiki/Air_quality_index#United_States) and [NowCast](https://en.wikipedia.org/wiki/NowCast_(air_quality_index)).

There's also IAQI (Indoor Air Quality Index) … which is like NowCast but designed for indoor air. It's like your "pm2.5 AQI" calculation but also a bunch of other readings that don't as much sense for outdoor readings. [Atmo has some info about the formula here](https://atmotube.com/atmocube-support/indoor-air-quality-index-iaqi). When I originally looked into this for my AirGradient DIY Pro, I think the issue was that the VOC and NOx readings weren't available in units compatible with the formula. However, it looks like Atmo has provided a table that could be used.

MallocArray, 2024-05-19:
Merged

closed 2024-05-19 by MallocArray