MallocArray/airgradient_esphome #47: Connection refused when "Upload to AirGradient Dashboard" is enabled

andyad8865, 2024-04-13:
Device info:
- airgradient
- by mallocarray
- Firmware: 2024.3.2 (Apr 13 2024, 14:10:11)
- Hardware: 2.0.1

this is related to #33 

Error:
`[15:22:58][W][http_request:087]: HTTP Request failed; URL: http://hw.airgradient.com/sensors/airgradient:<MAC-ADDR>/measures; Error: connection refused; Duration: 0 ms`

i have made sure it is added to the dashboard and the MAC address matches on both the URL in the error and the dashboard
[image]

other than that everything appears to work ok with the AG-ONE V9

MallocArray, 2024-04-13:
Are you setting a static IP in the YAML file, or letting it get one from DHCP?  If static, make sure you have a DNS server defined.

I have the same version of ESPHome working with my ONE, so for troubleshooting you may try using the official AirGradient firmware to see if it works.  That will confirm that the registration on the dashboard is correct.  (Double check that what you added to the dashboard is all lower case and matches what is shown on the ONE display when it first boots up)

andyad8865, 2024-04-13:
I have just installed the stock firmware from the airgradiant firmware and the dashboard shows the data for the AG-One:
[image]

andyad8865, 2024-04-13:
i'm going to now reinstall esphome, currently it runs in dhcp mode with the dns server set from the dhcp server

andyad8865, 2024-04-13:
Ah, I think I have found the issue, it appears it did not pick-up the DNS entries from the DHCP Server and had both DNS entries set at `0.0.0.0`, I have now added a static entry into the config and it is now reporting back.

`[18:43:41][D][http_request:099]: HTTP Request completed; URL: http://hw.airgradient.com/sensors/airgradient:<MAC-ADDR>/measures; Code: 200; Duration: 603 ms`

Sorry for the confusion, it appears I was a little blind when setting up the YAML file

closed 2024-04-13 by andyad8865

MallocArray, 2024-04-13:
Odd that it didn't grab DNS from DHCP, but glad you were able to nail it down.  This will be a  good reminder that if others get Connection Refused, to check DNS or add a manual entry.