MallocArray/airgradient_esphome #48: offset CO2 calibration based on updated outdoor ppm

DamonLane, 2024-04-14:
Since the SenseAir S8 calibration is apparently internally hardcoded to the ~2015 outdoor CO2 concentration of 400ppm, I think the package for that sensor should have an offset. The March 2024 number was 425. It varies by about 5 ppm between winter and summer and we don't need to get too precise about it, but 400 vs 425 feels big enough to correct for. It took almost 10 years for that change to happen, so updating every 3-10 years would be sufficient.

To implement this in my installation, I commented out the CO2 package and pasted the contents of it into my device yaml files, with the addition of the offset line.

packages/sensor_s8.yaml

```
sensor:
  - platform: senseair
    # SenseAir S8 https://esphome.io/components/sensor/senseair.html
    # https://senseair.com/products/size-counts/s8-lp/
    co2:
      name: "CO2"
      id: co2
      filters:
        - skip_initial: 2
        - offset: 25 
    id: senseair_s8
    uart_id: senseair_s8_uart

....[other parts of the package left untouched]...
```

MallocArray, 2024-04-15:
I think the best plan right now is to do exactly what you did, and manually add whatever offset you feel is appropriate.

CO2 concentrations change throughout the day and throughout wherever in the world you are, so there isn't a fixed number that is the "correct" offset.  Since the internal calibration number is 400, then all S8 sensors out there are calibrating to the same number, even if 400 isn't the exact measurement outdoors when the calibration happens.

You can see in this chart that the number fluctuates throughout the year.
https://en.wikipedia.org/wiki/Carbon_dioxide_in_Earth%27s_atmosphere#/media/File:Mauna_Loa_CO2_monthly_mean_concentration.svg

I'll add your link to the list of documentation for reference to get a current number, but that is for Mauna Loa Observatory and not necessarily your specific location.

MallocArray, 2024-05-13:
Addressed in commit 319c24520c371954c39f97f39b1f9c4bdcdf8fee

In your config file, you can now add a substitution that will let you offset the CO2 values reported

```yaml
substitutions:
  co2_offset: '0'
```

closed 2024-05-13 by MallocArray