MallocArray/airgradient_esphome #50: Temperature and Humidity go to "Unknown" with ESPHome version 2024.4.1

eatoff8, 2024-04-24:
I updated the ESPHome plug in, and then updated all the ESPHome devices on my network, and now the temperature and humidity go to "unknown" or "NaN"

I have tried both the sht3xd and sht4x platforms and neither work.

I am using the full config from here - https://github.com/MallocArray/airgradient_esphome/blob/main/full_config/ag-pro.yaml

When installing, this comes up in the log right at the very top, could it be related?

```
INFO ESPHome 2024.4.1
INFO Reading configuration /config/esphome/air-gradient.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing air-gradient (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
```

MallocArray, 2024-04-24:
I haven't tried 2024.4.1 yet.

Instead of the file in the full_config folder, you might try the one in the main folder. It mostly contains references to the packages in the repo, but represents the same thing, without possibly locking it to certain Arduino versions that may be updated.

I'll try to update my devices in a few days and see if I get anything similar.

I also suggest physically removing power to the device for 5 seconds and then plugging it back in. Sometimes that can clear weird things after and update that a software reset just won't resolve.

eatoff8, 2024-04-24:
> I'll try to update my devices in a few days and see if I get anything similar.

Thank you, much appreciated. I use the file from the full_config folder as I change the name of the device, and prefer to have a bit more control over the code.

MallocArray, 2024-04-24:
I pushed updated versions of the full_config and did see the versions change with 2024.4.1.

They compiled without error and are reporting on my side.

Michael-Phoenix, 2024-04-24:
Got the same problem since updating to 2024.4.1
And my temp + humidity still doesn't show with the latest version. :/
Got this in the startup log:
[19:23:55][C][sht3xd:038]: SHT3xD:
[19:23:55][E][sht3xd:040]:   Communication with SHT3xD failed!
[19:23:55][E][component:082]:   Component sht3xd.sensor is marked FAILED

MallocArray, 2024-04-24:
@Michael-Phoenix @eatoff8 
I'm not having any issues with the SHT4x devices, but after I updated the one device I have with SHT3x, I'm setting the same.

Looks like they are aware of the issue that started in 2024.4.0, but not fully resolved in 2024.4.1, so maybe next release?  In this thread there is a patch you can add an external_component to get it working now and remove when the final fix is in place, or go back to pre 2024.4.0 for now
https://github.com/esphome/issues/issues/5710#issuecomment-2063260398

Michael-Phoenix, 2024-04-24:
wow thank you for going on the hunt, I'll continue in that issue then

DariBer, 2024-04-26:
I'm not using the full file, I use the one that downloads each package, I have changed to use the sensor for temp_humidity sensor_sht30.yaml

I added at the end of the file the fix suggested above. But I still get the same error in the logs.

brentm5, 2024-04-29:
Mine also had the same issue.  It appears related to the following. 

https://github.com/esphome/issues/issues/5303

I added the i2c block from your repo [here](https://github.com/MallocArray/airgradient_esphome/blob/main/packages/airgradient_esp32-c3_board.yaml#L44-L48) into my config directly and changed the frequency back to 50khz.

```
i2c:
  sda: GPIO7 # Pin 21
  scl: GPIO6 # Pin 20
  frequency: 50kHz  # Changed the frequency back to the default and things started to work.
```

brentm5, 2024-04-30:
Just tested by removing the above and this appears to be fixed in [2024.4.2](https://github.com/esphome/esphome/releases/tag/2024.4.2) through the following PR https://github.com/esphome/esphome/pull/6592

MallocArray, 2024-04-30:
@brentm5 
Confirmed for me as well.  After upgrading to 2024.4.2 I was able to get it working, but I did have to fully power cycle the AirGradient Basic.  After doing an OTA install, I still had the same issue, but pulling the power cord for 10 seconds and then reconnecting got the temp/humidity working as expected.

eatoff8, 2024-05-02:
Thanks for the help, all working again with 2024.4.2 and the changes you made to the yaml