MallocArray/airgradient_esphome #52: AirGradient Open Air returns nothing but nan values for all sensors

ZogKarndon, 2024-05-05:
I've been running the ajfriesen firmware for my AirGradient Open Air sensors and decided to try your firmware, as it's been updated more recently.  

However, with the Open Air and the most recent firmware specs, I get nothing but NAN values from the Open Air, and Home Assistant shows nothing but Unknown values:
```
AirGradient Open Air AQI
Unknown
AirGradient Open Air AQI Category
Unknown
AirGradient Open Air NowCast
Unknown
AirGradient Open Air NowCast Category
Unknown
AirGradient Open Air Particulate Matter <1.0µm Concentration
Unknown
AirGradient Open Air Particulate Matter <10.0µm Concentration
Unknown
AirGradient Open Air Particulate Matter <2.5µm Concentration
Unknown
AirGradient Open Air Particulate Matter >0.3µm Count
Unknown
AirGradient Open Air Particulate Matter >0.5µm Count
Unknown
AirGradient Open Air Particulate Matter >1.0µm Count
Unknown
AirGradient Open Air Particulate Matter >2.5µm Count
Unknown
AirGradient Open Air Relative Humidity
Unknown
AirGradient Open Air Temperature
Unknown
AQI: Minutes Remaining
1,077 minutes
Button
Off
NowCast: Minutes Remaining
717 minutes
+20 entities not shown 
```
When I look at the EspHome logs, nothing obvious shows up, but all the sensors report NaN values.

MallocArray, 2024-05-05:
The names of the sensors are different between the configs, so if you have a card with the former sensors names, they will show nothing.

Try going to Settings>Devices and find your device to see all current sensors and hopefully you'll see the new sensors there with valid data.

If not, try removing power for a few minutes and then reconnect as that sometimes clears weird issues

MallocArray, 2024-05-06:
Were you able to get the new sensor names?

ZogKarndon, 2024-05-08:
Yes, I saw the new sensor names, but the values were still NaN.

MallocArray, 2024-05-08:
Does the display on the device show valid data? This should help clarify if the device isn't getting good values at all, or just not sending them to HA correctly.

If the display shows correct, then might want to try giving it a new name in the ESPHome config so it imports as a new device, or fully delete the device from HomeAssistant and re-add it.

ZogKarndon, 2024-05-08:
My OpenAir doesn't have a display (it's outside).   But the values coming from the logs are all NaN as well.

MallocArray, 2024-05-08:
My mistake for not re-reading this closely. 

I double checked ajfriesen's config and are using the same pins as my config:
https://github.com/ajfriesen/ESPHome-AirGradient/blob/main/air-gradient-open-air.yaml#L484

Is yours the PPT model with dual PM2.5 sensors or the PST that has one PM2.5 and one CO2 sensor?  Ensure you are using the correct config file.  I don't have a PPT myself, only PST, so if you have PPT, it is possible I missed something.

When you first start collecting logs wirelessly it should dump out a section in purple that shows what it detected at bootup.  Maybe something helpful in there. 

In the log output you included originally I didn't see anything about VOC or NOx, but maybe that wasn't included with the PPT model originally?

ex-nerd, 2024-05-15:
I have this occasionally happen with mine (and my DIY Pro models). This is usually fixed by a physical reboot (remove power for 5s or so).

Also, can confirm that VOC/NOx was not part of the original model, and that particular package needs to be commented out for our older device configs. At the time it was released, the add-on board was too big to fit into the enclosure and it wasn't even discussed as a possibility. He has since released a new smaller version of the add-on, but I only discovered this after poking around the AG forums/store when I noticed you had VOC referenced in your config.

See this post here: https://forum.airgradient.com/t/is-the-sgp41-tvoc-nox-compatible-with-the-outdoor-kit/1247

closed 2024-06-29 by MallocArray