MallocArray/airgradient_esphome #53: AirGradient ONE v9 PMS5003 sensor no values.

GeneralZero, 2024-05-08:
Device: AirGradient ONE v9 
Sensor: PMS 5003

All sensors other than the PMS5003 sensor work with the [airgradient-one.yaml](https://github.com/MallocArray/airgradient_esphome/blob/main/airgradient-one.yaml) file. I have reinstalled the [current firmware](https://www.airgradient.com/documentation/firmwares/) and checked that I am getting information from that sensor. 

From the android source code It looks like https://github.com/airgradienthq/arduino/blob/3efba24e24e956f12047b6d4832f20feb47cc495/src/Main/BoardDef.cpp#L165 uses UART0 to connect to that Sensor. 

What settings should be changed to test this?

saschaludwig, 2024-05-09:
> I have seen a few instances were after flashing, it is necessary to pull the power from the board entirely, as a software reboot hasn't been enough to get things going, but then it work as expected after that.

I had exactly this issue today. Just flashed my AG O1PST and I only got PM2.5 values from the PMS5003T.
After a power cycle, all values could be read. Thank you! :)

GeneralZero, 2024-05-11:
Weird I didn't change anything in the configuration regenerated the install file and now works. 
```
-a----          5/7/2024   4:42 PM        1305120 ag-one-factory (1).bin
-a----         5/11/2024   4:41 PM        1332368 ag-one-factory (2).bin
```

closed 2024-05-11 by GeneralZero