MallocArray/airgradient_esphome #55: Does this project support the previous 8266 based One?

Zipties, 2024-05-13:
I have the Airgradient-One version sold prior to the most current one with the LED strip. Would any of the configs be applicable for my model? I took a look at a couple of them and the 8266 based ones seemed to have different pinouts than what I'm using now with a custom esphome config.

MallocArray, 2024-05-13:
Sounds like you have the AirGradient Pro, since anything named `One` has the newer ESP32-C3 chip.

The file you'll want is `airgradient-pro.yaml`
If the temperature sensor in yours in the SHT30, you'll have to change one line in that file.  All of my Pros came with SHT40 which is the default in the config file

Zipties, 2024-05-15:
Ah, that must be in. My invoice just says "DIY Kit" so I couldn't remember exactly what they called it. I actually got it running with the basic config, and added in the VOC sensor. Seems to work, any reason I should change it?

MallocArray, 2024-05-15:
The Basic kit originally came with the SHT30 temp/hum sensor, while the current Pro kits come with SHT40.  If you are getting data from the temperature, then you are se there.

The display is also different between the two, and I wouldn't expect the Basic kit display package to work with the Pro, so I would expect to need to update that line.

If you are uploading to AirGradient's dashboard as well and added the VOC sensor, then you'll want the updated `airgradient_api` file that includes sending the VOC and NOx data

The later Pro models have a physical config button accessible through the back that you would need to add the `config_button` and `binary_sensor` sections.

So ultimately, if you have a Pro, I would sync up your config with the Pro config as far as the packages section goes to enable all functionality, but if you are happy where you are, you can stick with it

Zipties, 2024-05-15:
Thank you, Malloc. I'll get the config updated. I appreciate the feedback. You're right, the display is broken (I mostly forgot it existed), but the temperature and humidity work fine.

Zipties, 2024-05-15:
I saw that, so I had replaced them with my own values before flashing

`api:
  encryption:
    key: !secret home_assistant_encryption_key
ota:
  password: !secret ota_password
  safe_mode: true
  version: 2
  port: 8266
  reboot_timeout: 5min
  num_attempts: 10`

MallocArray, 2024-05-15:
Looks like a new issue with the 2024.5 release from today.  You could roll back or go without encryption until they fix it:
https://github.com/esphome/issues/issues/5790

closed 2024-05-19 by MallocArray