MallocArray/airgradient_esphome #56: senseair_s8_uart and pms5003_uart pins swapped on upgraded PPT model

alexbbt, 2024-05-14:
I am upgrading my PPT model to include the CO2 sensor. I found that using the PST config works, but only if I plug the sensors into the opposite ports.

The sensors should be plugged into these ports, for the case to fit correctly:
```
    - rx_pin: GPIO20  # Pin 30 or RX
      tx_pin: GPIO21  # Pin 31 or TX
      baud_rate: 9600
      id: senseair_s8_uart

    - rx_pin: GPIO0  # Pin 12
      tx_pin: GPIO1  # Pin 13
      baud_rate: 9600
      id: pms5003_uart
```

I am trying to manually make the YAML to test this. 

Similarly, I do not have a `tvoc` sensor since it was not included with my model, but using the PST config shows errors. Is there a way to disable this sensor?

alexbbt, 2024-05-14:
I was able to make my sensors work, in the plugs that fit, using the bellow config:
```
# AirGradient Open Air Outdoor Monitor with CO2
# Model: O-1PPT (with swapped PMS5003T sensor for CO2)
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  name: air-gradient-outdoor
  friendly_name: Air Gradient Outdoor
  config_version: 2.0.4
  name_add_mac_suffix: "false"  # Must have quotes around value

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-open-air-o-1pst.yaml
  import_full_config: false

packages:
  # board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
  safe_mode: github://MallocArray/airgradient_esphome/packages/switch_safe_mode.yaml

# Override for board package above
esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  name_add_mac_suffix: ${name_add_mac_suffix}  # Set to false if you don't want part of the MAC address in the name

  project:
    name: mallocarray.airgradient
    version: "$config_version"

  min_version: 2023.12.0

esp32:
  board: esp32-c3-devkitm-1

# esp32_improv:  # Uses around 30% of available flash space due to loading BLE stack
#   authorizer: none

# Enable logging
# https://esphome.io/components/logger.html
logger:
  baud_rate: 0  # Must disable serial logging as it conflicts with pms5003 uart pins and ESP32-C3 only has 2 hardware UART
  logs:
    component: ERROR  # Hiding warning messages about component taking a long time https://github.com/esphome/issues/issues/4717

# The captive portal is a fallback mechanism for when connecting to the configured WiFi fails.
# https://esphome.io/components/captive_portal.html
captive_portal:

# web_server:  # Please note that enabling this component will take up a lot of memory and may decrease stability, especially on ESP8266.

uart:
  # https://esphome.io/components/uart.html#uart
  - rx_pin: GPIO20  # Pin 30 or RX
    tx_pin: GPIO21  # Pin 31 or TX
    baud_rate: 9600
    id: senseair_s8_uart

  - rx_pin: GPIO0  # Pin 12
    tx_pin: GPIO1  # Pin 13
    baud_rate: 9600
    id: pms5003_uart

i2c:
  # https://esphome.io/components/i2c.html
  sda: GPIO7 # Pin 21
  scl: GPIO6 # Pin 20
  frequency: 400kHz  # 400kHz eliminates warnings about components taking a long time other than SGP40 component: https://github.com/esphome/issues/issues/4717

button:
  - platform: factory_reset
    disabled_by_default: true
    name: "Factory Reset ESP"
    id: factory_reset_all
```

MallocArray, 2024-05-14:
The config file was based on my assembled unit that I received configured as PST, so I think the stock hardware comes configured like the ESPHome file is configured.  I'm not able to find my photos of when I took mine apart right now.

What you did is perfectly valid though.  You commented out the package and included all of the same content, just changed which pins are used for each UART.

To remove the TVOC config, just comment out or remove the line that mentions the sgp41 sensor and it won't try to configure it.

MallocArray, 2024-05-18:
Another person mentioned they got cables of the same length and it sounds like that is a mistake.  If this is similar to you, contact their support e-mail

https://forum.airgradient.com/t/o-1pst-kit-jst-cables-both-the-same-length/2071/2

closed 2024-06-29 by MallocArray