MallocArray/airgradient_esphome #57: DIY Pro 4.2 needs option for sensor_sht30.yaml

ex-nerd, 2024-05-15:
Related to and maybe a solution for #19:

I have a couple DIY Pros with the 4.2 board and after switching to your config for the pro (https://github.com/MallocArray/airgradient_esphome/blob/nowcast_aqi/airgradient-pro.yaml) I noticed I was getting `nan` for temp/humidity. Going back to the other firmware, I realized that there must be an additional revision of the board because my devices need `sensor_sht30.yaml` not `sensor_sht40.yaml`. Making this change restores the temp/humidity readings

Anyway, I'm not sure how best you'd want to solve this (new board yaml for the pro-diy vs just a comment in the yaml that some boards use an older sensor) so I'm at least adding this here in case other users run into the same issue.

MallocArray, 2024-05-19:
I'll give it some additional thought.  I could add something to the Readme, or add a commented line in the config.  A similar ask has come up regarding the PMS5003 extended life configs and adding a commented line.  

Just not sure if I want to leave commented commands in the final file, but I do see that it is easier to know what you may need to do.

ex-nerd, 2024-05-19:
Comments seem reasonable for stuff like this. It's a pity that ESPHome doesn't (yet?) have support for fallback configs with hardware detection.

MallocArray, 2024-06-26:
Coming soon

MallocArray, 2024-07-12:
Added a commented line for SHT30 in the latest file for 4.0.0

closed 2024-07-12 by MallocArray