MallocArray/airgradient_esphome #62: Can you add a display*.yml option to display the time please?

letshin, 2024-05-28:
I think it would be useful for the airgradient to show the current time. Could you add a version displaying time please?

MallocArray, 2024-06-04:
I'll look at it, but not sure where it would fit into the standard AirGradient display page.  So it may need its own page

Adding the time option could have implications on memory usage that I would have to monitor.  

So it will be on the list to look at, but not saying it will be soon.

letshin, 2024-06-11:
Understood - perhaps an option to display text through a MQTT call then? That way users can add in what they want themselves on node red etc.

MallocArray, 2024-06-11:
I don't know how that would be accomplished, but if you figure something out, it can be considered as a new package that people could choose to add.
I haven't added the MQTT components to ESPHome and not sure of the impact to RAM and flash space, and not sure how to accept data that would then go into a page for the display.

letshin, 2024-06-11:
Got this working on an eink display so should be similar, not sure if it will work with airgradient yet as I haven't found time for that yet:
```
time:
  - platform: homeassistant
    id: homeassistant_time

text_sensor:
  - platform: template
    name: "Current Time"
    id: current_time
    update_interval: 10s
    lambda: return  id(homeassistant_time).now().strftime("%H%M");

display:
  - platform: waveshare_epaper
    id: eink_display
    ### truncated
    lambda: |-
      it.printf(240, 490, id(font3), TextAlign::TOP_CENTER, "%s", id(current_time).state.c_str());
```

MallocArray, 2024-09-07:
If someone else comes up with a complete file for it, it can be added as an optional package, but not on my list to work on at this time.

closed 2024-09-07 by MallocArray