MallocArray/airgradient_esphome #66: home assistant doesn't match air gradient dashboard (not enough decimals?(

HDCharles, 2024-06-18:
Air Gradient One home assistant values don't match what i see in the air gradient dashboard when they are very small

i.e.

[image]

it looks like its showing 0 decimal points because every once in a while PM2.5 shows 1 instead of 0

[image]

MallocArray, 2024-06-19:
I don't remember seeing decimal places in the past on the Dashboard so that may be a newer feature, or I'm mistaken

It looks like by default the ESPHome component sets it to accuracy of 0 digits of you don't specify it
https://github.com/MallocArray/airgradient_esphome/blob/dbdbcfea8dcc6ace47aeaaa57fdadbdfd3c6e32a/full_config/ag-one.yaml#L161

It could result in more database space used as it has to log more changes for any difference in decimal place instead of whole number. 

I can't test adding the accuracy change until next week but can explore it.

MallocArray, 2024-06-29:
I'm seeing inconsistencies in what is displayed on my side as well.  My Outdoor monitor does sometimes show a decimal place, but my indoor ones only show whole numbers when I've looked.  I wonder if it is using an average and getting decimals from that
https://forum.airgradient.com/t/dashboard-decimal-places/2236

I checked the API upload from the stock firmware and I only see whole numbers for PM values in it, so any decimals almost have to be related to an average calculated on the server side, but still checking

MallocArray, 2025-02-24:
There is some discussion here around decimal places, but they are talking about the HA integration with AirGradients using the stock firmware, so not fully related to ESPHome and what it is sending
https://forum.airgradient.com/t/ha-integration-decimal-places-in-pm2-5/3347/4

I'm pretty sure the sensor is only accurate to a whole number, so the stock firmware may be doing averages before sending, compared to this configuration that is just sending it as the sensor reports.

closed 2025-02-24 by MallocArray