MallocArray/airgradient_esphome #67: Error when building from esphome 2024.6 - "ESPHome supports certificate verification only via ESP-IDF"

prospartan2, 2024-06-20:
After modifying my airgradient-one.yaml with the required "platform: esphome" in the ota section (breaking change in esphome 2024.6) the build fails with the following log:
```

INFO ESPHome 2024.6.0
INFO Reading configuration /config/esphome/airgradient-one.yaml...
Failed config

http_request: [source /data/packages/76b4402f/packages/airgradient_api_esp32-c3.yaml:57]
  
  ESPHome supports certificate verification only via ESP-IDF. Set 'verify_ssl: false' to skip certificate validation and allow less secure HTTPS connections.
  {}
```
Here is my yaml minus the sensitive bits:
```
# AirGradient ONE - Board v9
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  name: "airgradient-one"
  friendly_name: "AirGradient One"
  config_version: 2.0.5
  name_add_mac_suffix: "false"  # Must have quotes around value

# Enable Home Assistant API
api:  # Add encryption key as desired
  encryption: 
    key: "********************************"

ota:  # Add password as desired
  platform: esphome
  password: "******************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: *********
    gateway: *********
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "AirGradientOne"
    password: "*************"

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-one.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  temp_humidity: github://MallocArray/airgradient_esphome/packages/sensor_sht40.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  display: github://MallocArray/airgradient_esphome/packages/display_sh1106_multi_page.yaml
  led: github://MallocArray/airgradient_esphome/packages/led.yaml
  led_co2: github://MallocArray/airgradient_esphome/packages/led_co2.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  config_button: github://MallocArray/airgradient_esphome/packages/config_button.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
  safe_mode: github://MallocArray/airgradient_esphome/packages/switch_safe_mode.yaml

binary_sensor:
  - id: !extend config_button
    pin:
      ignore_strapping_warning: true  # Acknowledging that this is a strapping pin and should not have external pullup/down resistors  https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
```

I assume this problem is coming from the changes to ota and http_request in the latest version of esphome.

idunnololz, 2024-06-20:
I am trying to setup a brand new AirGradient One and I am running into the same issue.

prospartan2, 2024-06-20:
If you comment out the airgradient_api package line it will build. I haven't tried updating my firmware with that work around yet.

MK00000000000000000000, 2024-06-20:
> This is probably not ideal, but adding the following to your yaml config file will get the build working:
> 
> ```
> http_request:
>   verify_ssl: false
> ```

Sorry, how do I downgrade? I'm using HAOS.

MK00000000000000000000, 2024-06-21:
> Lots of breaking changes in the latest release. I was reading the Release Notes last night and have an idea of the changes needed to get things compatible. I'll get to it this weekend.
> 
> Until then, you can downgrade ESPHome to 2024.5.x until then unless you want to find the exact changes needed and submit a PR or let me know

Sorry, how do I downgrade versions?

MallocArray, 2024-06-21:
> Sorry, how do I downgrade versions?

Looks like it isn't as easy as it could/should be, but this video has an option as well as another in the comments

https://www.reddit.com/r/Esphome/comments/1d2k9zl/how_to_downgrade_esphome_home_assistant_addon/

Motya-HA, 2024-06-21:
I have 45 ESP devices and I need to fix OTA line for everyone of them. Have 3 airgradients and still stuck with http_request compelling error

This is crazy, what these folks (ESPHome) were thinking??? New upgrades should be backward compatible. This is very frustrating

MallocArray, 2024-06-21:
I'm actively working on testing changes now, but I have it all working on my devices currently.  Here are the change:

```yaml
ota:  # Add password as desired
  platform: esphome

http_request:
  # Used to support POST request to send data to AirGradient
  # https://esphome.io/components/http_request.html
  verify_ssl: false  # Must be explicityly set to false when using Arduino framework
  # watchdog_timeout: 30s  # Slow response from API server can cause watchdog to reboot device
```

You can try adding/modifying these lines on your main YAML which should override the package contents until I get this pushed up to the repo.  I'm just making sure that it doens't break anything else.

The watchdog_timeout isn't tested yet, and will only work with the ESP32 boards (ONE and Open Air) but I haven't gotten to testing that yet.

MK00000000000000000000, 2024-06-21:
> I have 45 ESP devices and I need to fix OTA line for everyone of them. Have 3 airgradients and still stuck with http_request compelling error
> 
> This is crazy, what these folks (ESPHome) were thinking??? New upgrades should be backward compatible. This is very frustrating

I have less devices (8), same boat. I downgraded by adding this repo and installing esphome 5.5

Legacy ESPHome versions repository
https://github.com/khenderick/esphome-legacy-addons

...so now 6.1 is de-activated and stopped, this 5.5 version is running. Will be paying attention to when and what the real fix may be.

romines-dev, 2024-06-21:
> I'm actively working on testing changes now, but I have it all working on my devices currently. Here are the change:
>
> ```yaml
> ota:  # Add password as desired
>   platform: esphome
>
> http_request:
>   # [...]

Thanks for the quick patch! I've been able to update my devices when including this patch, and I'll also have one of my devices use that watchdog_timeout argument instead of the janky header patch from before to see if it works.

MallocArray, 2024-06-22:
I just pushed up changes to allow installing with 2024.6.0 and later.  This is the minimal changes needed. 

I have others that I'm still working on testing including the watchdog_timeout parameter, but standard install should now work.

cheezgr8r, 2024-06-23:
Thank you! Can confirm, this fixed my errors

MK00000000000000000000, 2024-06-23:
> I'm actively working on testing changes now, but I have it all working on my devices currently. Here are the change:
>
> ```yaml
> ota:  # Add password as desired
>   platform: esphome
>
> http_request:
>   # [...]

ditto, this worked for me. Thank you so much!

closed 2024-06-26 by MallocArray