MallocArray/airgradient_esphome #70: Bluetooth proxy for airgradient one (ESP32 C3)

Motya-HA, 2024-07-01:
I am trying to add Bluetooth proxy to airgradient one (ESP32 C3)

```
esp32_ble_tracker:
  id: ble_tracker
  scan_parameters:
    continuous: false

bluetooth_proxy:
  active: true
```

However, I am getting the following error:

```
Linking .pioenvs/guest-room-ag/firmware.elf
RAM:   [==        ]  17.8% (used 58356 bytes from 327680 bytes)
Error: The program size (1861062 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  101.4% (used 1861062 bytes from 1835008 bytes)
========================= [FAILED] Took 122.54 seconds =========================
```
I do not understand why bluetooth proxy lines can overflow flash

MallocArray, 2024-07-01:
By adding that feature, it has to load the entire bluetooth stack into the storage, and it ends up being more than this device can hold.  I saw similar things when looking to add the serial_improv feature.

It looks like it is just a bit over, so you could try removing things you don't need, like the Captive Portal, or if you are using the multi-page display config, switch to single.

If you aren't using the API upload to the AirGradient dashboard you could remove that as well.

raccettura, 2024-07-06:
I've been playing with this as well.  You need to use the `esp-idf` framework vs the default arduino framework to make it work, however switching the framework seems to cause some kind of reboot loop I can't quite figure out (my code is inspired by this repo, but not exactly the same thing).  I think the first step is figuring out how to run the `esp-idf` framework which per the [docs](https://esphome.io/components/bluetooth_proxy.html) saves significant memory.  Even for those not looking to use `bluetooth_proxy` some memory savings might be welcome to do other things that I can't even imagine ;)

nagisa, 2024-08-29:
FWIW here are some of the things I've done to get `esp-idf` going (if for no other reason but to get access to better WiFi options):

```
framework:
  type: esp-idf
  version: latest
  platform_version: 6.4.0
```

In particular this seems to have grabbed ESP-IDF version 5.1.2.

I also erased the flash with esptool.py before flashing a build with esp-idf, as I have experienced weird behaviour and crashes due to persistent data written by arduino firmware.

With this everything boots and runs without a hitch for me.

raccettura, 2024-09-05:
I switched over to this the other night.  As far as  I can tell it's reliable and heap size dropped dramatically even with the extra features added.  From about 190k B to 85k B.  That's a lot more headroom for... activities.

MallocArray, 2024-09-05:
Thanks for the work on this.  Since I'm not specifying the framework in any other package, I think it can just be added to your main yaml file for now.

I plan on testing it on a few of mine over the next few days to see how it goes.  Not sure if it will be a change to the existing package so all devices using this repo will use it, or if it would be an opt-in kind of thing where you specify a different package name to switch over.

MallocArray, 2024-09-06:
Successfully running.  I'm seeing mixed results in heap free.

Going to the latest IDF on my Outdoor model increased my heap free from 198k to 208k
[image]

Doing the same on my Indoor ONE, but setting the `platform_version: 6.4.0` went from 185k up to 192k Free for only a short while and then it rebooted and went back to 185k
[image]

raccettura, 2024-09-06:
[image]

here’s what I’m seeing, my firmware is technically more “inspired by” this repo than using it directly, but can’t think of any really huge difference.

MallocArray, 2024-09-06:
@raccettura I believe Heap Free is the amount of free memory, so the lower number would be less desirable, as you have less free.  Or am I misunderstanding that value?

I just switched my ONE to the `recommended` default version of esp-idf and my free went up to a similar number to the outdoor now, from 185k up to 197k.

So in my case, it looks like the default value for platform_version uses 5.4.0 and results is more Heap Free, while specifying 6.4.0 may not be a sustained increase, but I should test it again. I also see that version 6.8.1 has been released, so may try with some newer versions as well, unless you encountered problems that had you settle on 6.4.0
https://github.com/platformio/platform-espressif32/releases/

raccettura, 2024-09-06:
I think you are correct.  I'll also note by that screenshot I had enabled BT proxy by default, so really not an ideal comparison.

raccettura, 2024-12-24:
For anyone wondering:   I'm running the bluetooth proxy on two air gradients for a few months with no issues.  They work great.

Motya-HA, 2024-12-24:
I switched to MallocArray github YAML (no time to play with settings), so I am totally depend on him :-)
Would be happy when BT Proxy will be added to Airgradient
I am running several ESP32 BT proxies around house to maintain reliable BT connections. Having BT Proxy in Airgradient (I have them in every room) will let me drop all those ESP32 proxies

MallocArray, 2024-12-25:
I have also been trialing using the IDF and it was working well for me, less memory consumed. I was trying out different versions to get most memory and stability, and then had to back off for another issues. 

I plan on playing with it more over the next few days and likely to update the configs to use it by default soon. 

Appreciate the reminder here. 
I'm not sure that I'll include the Bluetooth proxy by default, but can add some instructions to add the proper lines to your individual files to enable or make it a separate package that can be added.

Motya-HA, 2024-12-25:
> I'm not sure that I'll include the Bluetooth proxy by default, but can add some instructions to add the proper lines to your individual files to enable or make it a separate package that can be added

> 

I think this is pretty straightforward. You can look at 

[https://github.com/EverythingSmartHome/everything-presence-lite](url)

Lewis did just additional packages with and without BLE

```
substitutions:
  name: "family-bar-epl-211588"
  friendly_name: Family Bar Presence LD2410
packages:

  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxx
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

I am using without BLE, because mmWave sensor is interfering with Bluetooth in the same sensor

MallocArray, 2025-01-13:
I pushed 4.1.0 of this config which defaults to the esp-idf framework for the ESP32 based devices.  This means you can add the bluetooth_tracker or esp32_improv options to your main config file if you desire.  

I haven't tried adding the bluetooth stuff yet, so not sure if it still fills up the available flash storage or not, but should make it at least an option for you to try out.

MallocArray, 2025-02-21:
Important note that with 4.1.0 and later, you need to do one USB flash of the devices, since OTA doesn't reformat the internal storage, and can result in you pushing a new update to it, then later the device rebooting and using the old partition, inadvertently downgrading your device.

I added a note about this in the README.

Another user mentioned not having enough storage to enable Bluetooth, but removing the captive_portal package was able to free up enough, and the portal shouldn't be needed in most circumstances once you have the device initially configured
https://github.com/MallocArray/airgradient_esphome/issues/115

closed 2025-02-21 by MallocArray