MallocArray/airgradient_esphome #71: ESP Home 2024.6.0 ota change

NathanTheGr8, 2024-07-09:
I am trying to use this to install an AG One with ESP 2024.6.0 on HA. I am getting errors when trying to prepare the config for install. 

I think the platform section requires 
`
ota:
  - platform: esphome
    password: "3f79ae5c6833318b1889bd8614246350"
`

like described [here](https://community.home-assistant.io/t/esphome-ota-configuration/730032/4).

After fixing that I get this error when preparing. Not sure what to do about it.
`
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/ag-one-basement.yaml...
Failed config

esphome: None
  name: ag-one-basement
  friendly_name: AG One Basement
  name_add_mac_suffix: False
  project: 
    name: mallocarray.airgradient
    version: 2.0.1
  min_version: 2023.12.0
  on_boot: 
    - priority: 200.0
      then: 
        - if: 
            condition: 
              switch.is_on: 
                id: upload_airgradient
            then: 
              - http_request.post: 
                  url: !lambda |-
                    return "http://hw.airgradient.com/sensors/airgradient:" + get_mac_address() + "/measures";
                  headers: 
                    Content-Type: application/json
                  json: 
                    wifi: !lambda |-
                      return to_string(-50);
                  
                  verify_ssl has moved to the base component configuration.
                  verify_ssl: True
                  method: POST
  build_path: build/ag-one
`

MallocArray, 2024-07-09:
Looks like you are using an older version of the config file and repo.  Both of these were addressed 2 weeks ago, so please compare your main config file to the one in this repo. It is labeled 2.0.5 but should have been incremented.

MallocArray, 2024-07-09:
Ah, that is on me. I pushed the new version of the individual files but hadn't regenerated a full config, since that happens after the other files are updated.

I'm just about to release a major new version and will take care of it then, as I don't think I can easily generate a new one with my in-progress work

MallocArray, 2024-07-12:
Updated full configs with latest changes

closed 2024-07-12 by MallocArray