MallocArray/airgradient_esphome #72: PM 2.5 can go negative in v4.0.0

joshschaeffer, 2024-07-14:
I am getting negative PM 2.5 readings from an outdoor O-1PST with the new v4.0.0 build. Seems to occur when the raw reading is bouncing between 0-1 with high humidity.

I am using the standard pms5003t package.

See attached HA screenshot.
[image]

joshschaeffer, 2024-07-15:
Nice! Thank you for keeping everything up to date. That correction formula seems like a step in the right direction, and is a big help.

MallocArray, 2024-07-15:
4.0.1 has been uploaded which should address this. I haven't encountered negative numbers myself, so only assuming that the changes are addressing it.

closed 2024-07-15 by MallocArray