MallocArray/airgradient_esphome #73: looks like a substitution, but 'friendly_name' was not declared

bmos, 2024-07-25:
Validation output
```console
INFO ESPHome 2024.7.1
INFO Reading configuration /config/esphome/living-room-airgradient.yaml...
WARNING Found 'it.printf(0, 0, id(open_sans_14), "ID:");
it.printf(128, 0, id(open_sans_14), TextAlign::TOP_RIGHT, "%s", get_mac_address().c_str());
it.printf(0, 21, id(open_sans_14), "Config Ver: $config_version");
it.printf(0, 42, id(open_sans_14), "$friendly_name");' (see display->0->pages->1->lambda) which looks like a substitution, but 'friendly_name' was not declared
WARNING Found 'it.printf(0, 0, id(open_sans_14), "ID:");
it.printf(128, 0, id(open_sans_14), TextAlign::TOP_RIGHT, "%s", get_mac_address().c_str());
it.printf(0, 21, id(open_sans_14), "Config Ver: 2.0.1");
it.printf(0, 42, id(open_sans_14), "$friendly_name");' (see display->0->pages->1->lambda) which looks like a substitution, but 'friendly_name' was not declared
```

```yaml
substitutions:
  config_version: 2.0.1
  co2_offset: '0'
  voc_learning_time_offset_hours: '12'
  nox_learning_time_offset_hours: '12'
  co2_green: '400'
  co2_yellow: '1000'
  co2_red: '2000'
  co2_purple: '4000'
  name_add_mac_suffix: 'false'
esphome:
  name: living-room-airgradient
  friendly_name: Living Room AirGradient
  name_add_mac_suffix: false
  project:
    name: mallocarray.airgradient
    version: 2.0.1
  min_version: 2024.6.0
[...]
```

am I doing something wrong?
here is my yaml:

```yml
# AirGradient ONE - Board v9
# https://www.airgradient.com/open-airgradient/instructions/overview/

esphome:
  name: "living-room-airgradient"
  friendly_name: "Living Room AirGradient"
  name_add_mac_suffix: "false"  # Must have quotes around value

http_request:
  verify_ssl: false

substitutions:
  config_version: 2.0.1
  name_add_mac_suffix: "false"

# Enable Home Assistant API
api:
  encryption:
    key: "gezhvrehvehenernerbvcer/gezhvrehvehenernerbvcer/gezhvrehvehenernerbvcer="

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret ap_password

ota:
  - platform: esphome
    password: "gezhvrehvehenernerbvcer"

captive_portal:

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-one.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  captive_portal: github://MallocArray/airgradient_esphome/packages/captive_portal.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  temp_humidity: github://MallocArray/airgradient_esphome/packages/sensor_sht40.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  display: github://MallocArray/airgradient_esphome/packages/display_sh1106_multi_page.yaml
  led: github://MallocArray/airgradient_esphome/packages/led.yaml
  led_co2: github://MallocArray/airgradient_esphome/packages/led_co2.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  config_button: github://MallocArray/airgradient_esphome/packages/config_button.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml

binary_sensor:
  - id: !extend config_button
    pin:
      ignore_strapping_warning: true  # Acknowledging that this is a strapping pin and should not have external pullup/down resistors  https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
```

bmos, 2024-07-25:
I figured it out.
new yaml:
```yaml
# AirGradient ONE - Board v9
# https://www.airgradient.com/open-airgradient/instructions/overview/

# These substitutions allow the end user to override certain values
substitutions:
  name: "living-room-airgradient"
  friendly_name: "Living Room AirGradient"
  config_version: 2.0.1

esphome:
  name: "${name}"
  # Friendly names are used where appropriate in Home Assistant
  friendly_name: "${friendly_name}"
  name_add_mac_suffix: "false"  # Must have quotes around value

http_request:
  verify_ssl: false
[...]
```

closed 2024-07-25 by bmos

MallocArray, 2024-07-25:
I've modified the layout of the files considerably from when you originally created your file, so you may want to consider updating it to follow the most current layout, which would also have the modifications you made.

```yaml
# AirGradient ONE - Board v9
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  name: "living-room-airgradient"
  friendly_name: "Living Room AirGradient"
  name_add_mac_suffix: "false"  # Must have quotes around value

api:
  encryption:
    key: "gezhvrehvehenernerbvcer/gezhvrehvehenernerbvcer/gezhvrehvehenernerbvcer="

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret ap_password

ota:
  - platform: esphome
    password: "gezhvrehvehenernerbvcer"

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-one.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  captive_portal: github://MallocArray/airgradient_esphome/packages/captive_portal.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  temp_humidity: github://MallocArray/airgradient_esphome/packages/sensor_sht40.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  display: github://MallocArray/airgradient_esphome/packages/display_sh1106_multi_page.yaml
  led: github://MallocArray/airgradient_esphome/packages/led.yaml
  led_co2: github://MallocArray/airgradient_esphome/packages/led_co2.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  config_button: github://MallocArray/airgradient_esphome/packages/config_button.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml

binary_sensor:
  - id: !extend config_button
    pin:
      ignore_strapping_warning: true  # Acknowledging that this is a strapping pin and should not have external pullup/down resistors  https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
```