MallocArray/airgradient_esphome #74: Update AQI breakpoints

richli, 2024-07-27:
I noticed that [recently the US EPA updated the AQI formula](https://www.epa.gov/system/files/documents/2024-02/pm-naaqs-air-quality-index-fact-sheet.pdf). They changed a few of the breakpoints, such as lowering the first PM 2.5 bin from 12 μg/m³ to 9 μg/m³.

The AQI calculation in files such as `sensor_pms5003.yaml` cites [the Wikipedia page for AQI](https://en.wikipedia.org/wiki/Air_quality_index#Computing_the_AQI), and it has already been updated with the changes.

MallocArray, 2024-08-01:
Thank you.  I wasn't aware of the update.  

Looking at it.

MallocArray, 2024-08-01:
Testing now
https://github.com/MallocArray/airgradient_esphome/pull/76

MallocArray, 2024-09-05:
Included in the 4.0.5 release

closed 2024-09-05 by MallocArray