MallocArray/airgradient_esphome #79: watchdog name conflict prevents update

domoritz, 2024-08-21:
I cannot update to ESPHome 2024.8.0 with the default config because of this error

`ID 'watchdog' conflicts with the name of an esphome integration, please use another ID name.`

MallocArray, 2024-08-21:
There was something in the change log about separating out the watchdog component so it wasn't linked only to the http_request component.  I'll take a look at what needs to be done.

MallocArray, 2024-08-21:
Just pushed an update to the repo.  If you have already tried to update today, ESPHome by default keeps a cache of packages for 24 hours, so you may not be able to pull it down until that expires, unless you can delete the `config/.esphome/packages/` directory.

closed 2024-08-22 by MallocArray