MallocArray/airgradient_esphome #80: Compile error with ESPHome 2024.8.0 for

jeroenleenarts, 2024-08-21:
```
INFO ESPHome 2024.8.0
INFO Reading configuration /config/esphome/air-gradient-pro.yaml...
Failed config

interval: [source /data/packages/76b4402f/packages/display_sh1106_multi_page.yaml:192]
  - interval: 10s
    startup_delay: 1s
    then: 
      if: 
        condition: 
          switch.is_on: display_boot_page
        then: 
          switch.turn_off: display_boot_page
  - interval: 5s
    then: 
      - if: 
          condition: 
            switch.is_on: display_blank_page
          then: 
            - display.page.show: blank
            - component.update: oled_display
          else: 
            - display.page.show_next: oled_display
            - component.update: oled_display
  - interval: 5s
    then: 
      - if: 
          condition: 
            light.is_on: led_strip
          then: 
            - if: 
                condition: 
                  lambda: return !(isnan(id(co2).state));
                then: 
                  - light.addressable_set: 
                      id: led_strip
                      color_brightness: !lambda |-
                        return ((id(led_brightness).state - (id(led_fade).state * 5)) / 100.0);
                      range_from: 0
                      range_to: 10
                      red: !lambda |-
                        return (id(co2).state < 400) ? 0 : (id(co2).state <= 1000) ? (id(co2).state - 400) / (1000 - 400) : 1;
                      green: !lambda |-
                        return (id(co2).state < 1000) ? 1 : (id(co2).state <= 2000) ? 1 - (id(co2).state - 1000) / (2000 - 1000) : 0;
                      blue: !lambda |-
                        return (id(co2).state < 2000) ? 0 : (id(co2).state <= 4000) ? (id(co2).state - 2000) / (4000 - 2000) : 1;
                  - light.addressable_set: 
                      id: led_strip
                      color_brightness: !lambda |-
                        return ((id(led_brightness).state - (id(led_fade).state * 4)) / 100.0);
                      range_from: 1
                      range_to: 9
                      red: !lambda |-
                        return (id(co2).state < 400) ? 0 : (id(co2).state <= 1000) ? (id(co2).state - 400) / (1000 - 400) : 1;
                      green: !lambda |-
                        return (id(co2).state < 1000) ? 1 : (id(co2).state <= 2000) ? 1 - (id(co2).state - 1000) / (2000 - 1000) : 0;
                      blue: !lambda |-
                        return (id(co2).state < 2000) ? 0 : (id(co2).state <= 4000) ? (id(co2).state - 2000) / (4000 - 2000) : 1;
                  - light.addressable_set: 
                      id: led_strip
                      color_brightness: !lambda |-
                        return ((id(led_brightness).state - (id(led_fade).state * 3)) / 100.0);
                      range_from: 2
                      range_to: 8
                      red: !lambda |-
                        return (id(co2).state < 400) ? 0 : (id(co2).state <= 1000) ? (id(co2).state - 400) / (1000 - 400) : 1;
                      green: !lambda |-
                        return (id(co2).state < 1000) ? 1 : (id(co2).state <= 2000) ? 1 - (id(co2).state - 1000) / (2000 - 1000) : 0;
                      blue: !lambda |-
[... 119 lines cut ...]

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-one.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  temp_humidity: github://MallocArray/airgradient_esphome/packages/sensor_sht40.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  display: github://MallocArray/airgradient_esphome/packages/display_sh1106_multi_page.yaml
  led: github://MallocArray/airgradient_esphome/packages/led.yaml
  led_co2: github://MallocArray/airgradient_esphome/packages/led_co2.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  // hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  config_button: github://MallocArray/airgradient_esphome/packages/config_button.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml

binary_sensor:
  - id: !extend config_button
    pin:
      ignore_strapping_warning: true  # Acknowledging that this is a strapping pin and should not have external pullup/down resistors  https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
```

jeroenleenarts, 2024-08-21:
As you can see. I commented the 'hardware_watchdog' package, but that didn't help.

MallocArray, 2024-08-21:
There was something in the change log about separating out the watchdog component so it wasn't linked only to the http_request component.  I'll take a look at what needs to be done.

def1149, 2024-08-21:
I experienced the same issue. 
It looks like the ID watchdog is now taken by some other ESPHome code

MallocArray, 2024-08-21:
> As you can see. I commented the 'hardware_watchdog' package, but that didn't help.

I'm surprised that commenting out the watchdog package didn't resolve it.

MallocArray, 2024-08-21:
> I experienced the same issue. It looks like the ID watchdog is now taken by some other ESPHome code Maybe you just need to rename your ID to a1_watchdog

I just updated watchdog.yaml to the following which is letting me get past the previous hurdle.  Doing some additional testing before pushing it up to the repo.  If you wanted to also test, you could comment out the watchdog.yaml line and add this at the bottom of your file:

```yaml
output:
  - platform: gpio
    # Pin to notify external watchdog that activity is present
    id: external_watchdog
    pin:
      number: GPIO2
      ignore_strapping_warning: true  # Acknowledging that this is a strapping pin and should not have external pullup/down resistors  https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins

interval:
  - interval: 2.5min
    # Notify watchdog device it is still alive
    then:
      - output.turn_on: external_watchdog
      - delay: 20ms
      - output.turn_off: external_watchdog
```

jeroenleenarts, 2024-08-21:
Got EspHome running as an add on. So I will take the 24h wait. 👍

def1149, 2024-08-22:
Update Successful. No errors. The new firmware executing and reporting to Home Assistant 
Thanks

closed 2024-08-22 by MallocArray

jeroenleenarts, 2024-08-22:
Fixed