MallocArray/airgradient_esphome #83: AG Pro resetting every ~2 minutes

sleien, 2024-09-03:
Hi,

Since the update my AG Pro (sht30) keeps resetting every ~2 minutes with the ESPHome logs being:

```
...
[23:12:17][D][pmsx003:234]: Got PM1.0 Concentration: 8 µg/m^3, PM2.5 Concentration 14 µg/m^3, PM10.0 Concentration: 14 µg/m^3
[23:12:17][D][sensor:093]: 'PM 1.0': Sending state 8.00000 µg/m³ with 0 decimals of accuracy
[23:12:17][D][sensor:093]: 'PM 2.5': Sending state 8.20784 µg/m³ with 0 decimals of accuracy
[23:12:17][D][sensor:093]: 'PM 10.0': Sending state 14.00000 µg/m³ with 0 decimals of accuracy
[23:12:17][D][sensor:093]: 'PM 0.3': Sending state 1548.00000 /dL with 0 decimals of accuracy
[23:12:17][D][sgp4x:276]: Sensor has not collected enough samples yet. (89/90) VOC index is: 97
[23:12:18][D][sgp4x:276]: Sensor has not collected enough samples yet. (90/90) VOC index is: 98
WARNING ag-pro @ 192.168.1.52: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for ag-pro @ 192.168.1.52
WARNING Disconnected from API
INFO Successfully connected to ag-pro @ 192.168.1.52 in 1.024s
INFO Successful handshake with ag-pro @ 192.168.1.52 in 0.107s
[23:12:49][C][uptime.sensor:033]: Uptime Sensor 'Uptime'
[23:12:49][C][uptime.sensor:033]:   Device Class: 'duration'
[23:12:49][C][uptime.sensor:033]:   State Class: 'total_increasing'
[23:12:49][C][uptime.sensor:033]:   Unit of Measurement: 's'
[23:12:49][C][uptime.sensor:033]:   Accuracy Decimals: 0
...
```

After the resets the screen shows nan for a while and then displays the correct values again. With the AG firmware there was no problem. It's in the same wifi as before.

Also I activated the send to AG Dashboard but on my dashboard the device is greyed out and it seems there is no data received (maybe because of the resets).

MallocArray, 2024-09-03:
Thanks.  I have 3 different Pro devices with various sensor combinations and boards and one of them has been doing a similar thing. It appears to be related to switching to HTTPS communication to the AG Dashboard.  I've just been needing a push to go make a second package that has the insecure method that has been in use until just recently.

Until then, if you want to remove or comment out the airgradient_api_* line and try to install it again and see if that stabilizes it.  If so, that will help solidify that is the issue.  You can also try turning off the AG Dashboard option in HomeAssistant, but not sure if that will take if it is the same issue I'm seeing on one device.

I'll work on making the insecure package in the next day or two and update this Issue when completed

sleien, 2024-09-04:
This seems to work.

There are still some errors and warning about SenseAir in the logs but it doesn't seem to impact the AG.

```
[09:25:51][E][uart:015]: Reading from UART timed out at byte 7!
[09:25:51][W][senseair:024]: Reading data from SenseAir failed!
...
[09:26:51][W][senseair:030]: Invalid preamble from SenseAir! 38800760 01ae78fe 04080000 00
```

MallocArray, 2024-09-05:
I just added the new package and my device that was rebooting has been up for 15 minutes now.

Change the airgradient_api line to this:
`airgradient_api: !include packages/airgradient_api_d1_mini_insecure.yaml`

This will change the API communication from the newer HTTPS:// setting to the insecure and original HTTP:// method

closed 2024-09-07 by MallocArray

sleien, 2024-09-09:
The fix works so it's not crashing anymore but the switch is not appearing on the dashboard. Is this because it has the same id as the switch in the secure package?

MallocArray, 2024-09-09:
It should be fine, as in the end, it is the same switch and functionality, so it shouldn't know about any difference.  You changed the existing line and didn't add both the original and insecure, right?

Maybe refresh pages or try a different browser. I just checked mine and the Upload switch is still there

sleien, 2024-09-09:
Yes I commented the secure one out and inserted the insecure.

Also with a different browser and multiple reinstalls with different orders of the packages this happens. When I'm looking at the 23 entities of the devices it also doesn't show anything with upload. 

Entities shown:
AG Pro CO2
sensor.ag_pro_co2
—
AG Pro Display AirGradient Default Page
switch.ag_pro_display_airgradient_default_page
—
AG Pro Display Blank Page
switch.ag_pro_display_blank_page
—
AG Pro Display Boot Page
switch.ag_pro_display_boot_page
—
AG Pro Display Contrast %
number.ag_pro_display_contrast
—
AG Pro Display Temperature in °F
switch.ag_pro_display_temperature_in_f
—
AG Pro Firmware
update.ag_pro_firmware
—
AG Pro Humidity
sensor.ag_pro_humidity
—
AG Pro NOx Index
sensor.ag_pro_nox_index
—
AG Pro PM 0.3
sensor.ag_pro_pm_0_3
—
AG Pro PM 1.0
sensor.ag_pro_pm_1_0
—
AG Pro PM 10.0
sensor.ag_pro_pm_10_0
—
AG Pro PM 2.5
sensor.ag_pro_pm_2_5
—
AG Pro SenseAir S8 Automatic Baseline Correction
switch.ag_pro_senseair_s8_automatic_baseline_correction
—
AG Pro SenseAir S8 Calibration
button.ag_pro_senseair_s8_calibration
—
AG Pro SenseAir S8 Show Calibration Interval
button.ag_pro_senseair_s8_show_calibration_interval
—
AG Pro Temperature
sensor.ag_pro_temperature
—
AG Pro Uptime
sensor.ag_pro_uptime
—
AG Pro VOC Index
sensor.ag_pro_voc_index
—
AG Pro WiFi Signal
sensor.ag_pro_wifi_signal
—
Humidity Raw
sensor.ag_pro_humidity_raw
—
PM 2.5 AQI
sensor.ag_pro_pm_2_5_aqi
—
Temperature Raw
sensor.ag_pro_temperature_raw

MallocArray, 2024-09-09:
That is odd.  You could try doing another install from ESPHome, as I've seen some weird things in the past that a reinstall fixed, like my display being upside down.  

You could also try adding this package and reinstalling:
`factory_reset: github://MallocArray/airgradient_esphome/packages/button_factory_reset.yaml`

You should get a new button for Factory Reset that will erase all saved options like which switches you have flipped from default and LED or display brightness, but will keep your configured wifi as long as that is in your YAML file.  It could also restore some behavior.