MallocArray/airgradient_esphome #97: Some data missing @ DIY PRO version 4.2

YunsuChoi, 2024-10-12:
Hello. I've applied your 4.0.7 release to my v4.2 board. When I using airgradient's software every sensor data was visible but 4.0.7 isn't.
what did only complied airgradient-pro.yaml with wifi SSID and PASSWD. 
and here is the result. 
please review the attachment. Thanks. 
[image]
[image] 
 - 알 수 없음 means Unknown. in Korean.

MallocArray, 2024-10-12:
I haven't seen that before. Since the display isn't showing values, this isnt a Home assistant specific issue. 

I would try removing the power cable for 10 seconds and then connect it again. I've seen some odd issues that require more than just a software reset.

putz612, 2024-10-12:
I am having issues as well, it seems to get into a boot loop.  I have tired factory resets but it still happens where it just says nan on a few of the sensors.  If I watch it enough I will see the boot message again.

MallocArray, 2024-10-12:
If it looks like the while device is rebooting try either disabling the AirGradient Upload switch in Home Assistant, or comment out the API package and reinstall. 

When the server is having issues, it can cause the devices to reboot regularly

putz612, 2024-10-12:
I have been running without the API package enabled.

MallocArray, 2024-10-12:
Interesting. I'll try the latest esphome with my devices. I havent done the last minor patch

YunsuChoi, 2024-10-14:
I've flahsing compiled full-config yaml version but it displayed version 4.0.6. seems odd. please check release package itself.

MallocArray, 2024-10-14:
@YunsuChoi I updated the full-config folder with the latest yaml files

MallocArray, 2024-10-14:
@putz612 
I just checked and my install on a Pro 4.2 is on the latest generally available build 2024.9.2, so not sure why you are getting the boot loop, especially without the API package. 
If you want to try you could add another package
`factory_reset: github://MallocArray/airgradient_esphome/packages/button_factory_reset.yaml`
which will add a button to do a Factory Reset, which sounds more scary than it is.  It will remove all of your settings you changed on your device compared to the defaults, but doesn't fully erase your ESP device.  Sometimes I've gotten messages that I ran out of space in the nvram and have to do this.  So if you changed from F to C for temperature, it would reset this back to the default.

YunsuChoi, 2024-10-14:
> @YunsuChoi I updated the full-config folder with the latest yaml files

1. Everytime I flash the release, I erase all via web.esphome.io. therefore sort of factory reset has been done.
2. Okay then, Should I try from git master branch full-config>ag-pro.yaml?

YunsuChoi, 2024-10-14:
> @putz612 I just checked and my install on a Pro 4.2 is on the latest generally available build 2024.9.2, so not sure why you are getting the boot loop, especially without the API package. If you [...]

1. Everytime I flash the release, I erase all via web.esphome.io. therefore sort of factory reset has been done.
2. Okay then, Should I try from git master branch full-config>ag-pro.yaml?

YunsuChoi, 2024-10-14:
> @YunsuChoi I updated the full-config folder with the latest yaml files

Okay then, Should I try from git main branch > full_config > ag-pro.yaml?

MallocArray, 2024-10-15:
@YunsuChoi 
It all depends on how you are running your personal setup.
My recommendation would be to have ESPHome installed locally and use the `ag-pro.yaml` file from the main branch main folder as the content of your personal device config. You can change the substitutions to make the name what you want, add your own encryption keys or wifi settings, etc.  To update, you would just do an Install from your ESPHome and it should pull the latest config from this repo and send to your personal device.

Option 2 is to use the file from the `full_config` folder, but you won't get the updates regularly from there, since it is completely standalone and doesn't refer back to this repo, so updating as changes are made here become a more manual process

Option 3 is to install the .bin files manually which doesn't let you configure anything yourself since they are already compiled as they are configured in this repo and you can't update without having to manually install the latest .bin, but this option doesn't require that you have ESPHome in your own environment.

So the option is up to you as to how you want to do it, but there are considerations.

That being said, I just updated the .bin files with the latest config.  I still need to automate that process and I have to manually generate them at this time and I didn't do it after the last update.

YunsuChoi, 2024-10-16:
> @YunsuChoi It all depends on how you are running your personal setup. My recommendation would be to have ESPHome installed locally and use the `ag-pro.yaml` file from the main branch main folder as [...]

Thanks for detailed manual. however some of my module(SGP41) is broken. So I think real world testing is going to be quite delayed. Thanks again.

YunsuChoi, 2024-10-24:
MallocArray, finally I fix proper SGP41 module and flash your 4.0.7 binary([airgradient-pro.bin](https://github.com/MallocArray/airgradient_esphome/blob/main/airgradient-pro.bin)), but it gets worse. none of sensors are displayed. Just 'nan'.
Please check your 4.0.7 build code. cause it shows every elements while in Airgradient's software.

MallocArray, 2024-10-24:
I'm not sure why it isn't working on yours.  I have a 4.2 board with stock sensors and it is working with this code for me.  You might try disabling the switch `Upload to AirGradient Dashboard` in the Device Options of HomeAssistant so it doesn't report to the API which can cause some issues.

You can also try changing the package for the API to the insecure method
`airgradient_api: !include packages/airgradient_api_d1_mini_insecure.yaml`
This doesn't use HTTPS which AirGradient didn't do for a very long time and I'm not sure if the current stock firmware is using it.
On a 3.7 board with a D1 Mini I had to use this insecure option, but on my 4.2 with the same D1 Mini I could use the regular one that is now using HTTPS

Give one or both of those a try and see what happens.  If it turns out that other D1 Mini based devices continue having trouble when attempting to use HTTPS, I may need to make the default be insecure

MallocArray, 2025-01-13:
Are you still having issues?

YunsuChoi, 2025-01-13:
Thanks for concern but I gave up this issue 🙃 
Thanks again.

closed 2025-01-13 by MallocArray

putz612, 2025-01-16:
I did figure out a solution to my issues with the reboots.  I believe the esp8266 that comes with the original DIY pro is a bit under powered so we hit a watchdog timer and it reboots.  Thankfully there is a drop in pin compatible esp32.

https://www.amazon.com/gp/product/B08L79192D
NOTE: there are newer ones that have USB-C now but this is what I had sitting on my bench.

I took the airgradient_d1_mini_board.yaml and replaced

```
esp8266:
  board: d1_mini
  restore_from_flash: true  # Used to store state of display config
```
with
```
esp32:
  board: wemos_d1_mini32
```

The pins are identical as long as you use the ones that have the white silk screen on the PCB.   Not sure if this would be worth adding in as a file or putting in the README but I have had zero issues since I did this.  

I wonder if this might help #106 and others.