MallocArray/airgradient_esphome #102: AG Pro without sgp4x sensor for VOC and NOx

bjoernmayer, 2024-10-26:
Heyho,

early shipments of the AG pro did not include the sensor for VOC and NOx.
On these devices the `ag-pro` config will show `nan` for both values in the third column

MallocArray, 2024-10-27:
That would be expected since the sensors aren't there.

Some options today are to switch from the single_page display to the multi_page display package and only turn on the pages that support the sensors you have.

It would be possible to make a version of the single page that leaves out the third column and spaces the other components out wider, but honestly, it takes a lot of time to trial and error how the spacing should be, and I don't expect to invest the 3-5 hours it would likely take to test it.

If someone comes up with a suitable config file, I would be interested in adding it as an option to the available packages.

bjoernmayer, 2024-10-28:
Would you rather have another package for a 2 column display, or add a condition in the current single page display package? Something like `if (hasSGP4X) // 3 cols else // 2 cols`

MallocArray, 2024-10-28:
I don't think something like that is possible with ESPHome, as far as detecting if it has a sensor installed or not.
I'm actually surprised that without the VOC sensor, it doesn't throw a larger error that keeps it from working at all.

I suppose you could do an IF check in the code for the display that says if the VOC sensor is not NaN, then do the 3 column code, else do the 2 column one, but that eats more memory and RAM that the D1 Mini is just struggling with all of this.

So if you are interested in working on it, I would recommend first making a new package just for the 2 column display, based on the current single_page file, but laid out for 2 columns.

If you get that and we can test it out, we can add it as a separate packgae, or then try it as a giant Else block in the current file and see what the impact to storage and RAM is, to see if it is worth it.

For what it is worth, my 4.2 board with ESPHome has 13.5k free RAM and once it goes below about 10k, it starts crashing/rebooting:

[image]

closed 2025-02-21 by MallocArray