MallocArray/airgradient_esphome #103: Feature Request: Add clock between temp and humid

bjoernmayer, 2024-10-26:
Hey,

would it be possible to show the current time in between the current temperature and the humidity?

MallocArray, 2024-10-26:
It has been requested in the past.  The space is pretty cramped and it would require adding more packages to handle time sync and at least for the Pro and earlier models, I think it would use up the remaining resources of that chip.

It could be added to the ONE model possibly, but nothing I'm planning on investigating at this time.
If someone else comes up with the required changes for it, I would be interested in making it available as an optional package.

bjoernmayer, 2024-10-28:
Alright. Thank you for your response

closed 2025-03-21 by MallocArray