MallocArray/airgradient_esphome #105: Fails to compile in ESPHome 11.0+

Ragingfire105, 2024-11-23:
I have a mostly stock configuration for a AG One.  After updating to ESPHome 11.0 it was not able to compile with the below error. Most other ESPs did update successfully.

```
INFO ESPHome 2024.11.1
INFO Reading configuration /config/esphome/airgradient-one.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 503, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 212, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 224, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 684, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 204, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/font/__init__.py", line 557, in to_code
    base_font = EFont(config[CONF_FILE], size, point_set)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/font/__init__.py", line 475, in __init__
    self.font = load_ttf_font(path, size)
                ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/font/__init__.py", line 519, in load_ttf_font
    from PIL import ImageFont
  File "/usr/local/lib/python3.11/dist-packages/PIL/ImageFont.py", line 39, in <module>
    from . import Image
  File "/usr/local/lib/python3.11/dist-packages/PIL/Image.py", line 100, in <module>
    from . import _imaging as core
ImportError: libtiff.so.6: cannot open shared object file: No such file or directory
```

Congfig Yaml:
```
# AirGradient ONE - Board v9
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  name: "ag-one"
  friendly_name: "AG One"
  config_version: 2.0.1
  name_add_mac_suffix: "false"  # Must have quotes around value

# Enable Home Assistant API
api:  # Add encryption key as desired

ota: 
  platform: esphome

wifi:
 ##Redacted##
dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-one.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  temp_humidity: github://MallocArray/airgradient_esphome/packages/sensor_sht40.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  display: github://MallocArray/airgradient_esphome/packages/display_sh1106_multi_page.yaml
  led: github://MallocArray/airgradient_esphome/packages/led.yaml
  led_combo: github://MallocArray/airgradient_esphome/packages/led_combo.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  config_button: github://MallocArray/airgradient_esphome/packages/config_button.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
  safe_mode: github://MallocArray/airgradient_esphome/packages/switch_safe_mode.yaml

binary_sensor:
  - id: !extend config_button
    pin:
      ignore_strapping_warning: true  # Acknowledging that this is a strapping pin and should not have external pullup/down resistors  https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins

```

MallocArray, 2024-11-23:
On the whole things look good.
Do please remove the line `config_version: 2.0.1` as that was moved into the board package so it gets updated as the files in the project get updated, but I don't expect that to cause your issue.

This looks mostly like an ESPHome cached file issue.  I have things running on 2024.11.0 successfully, so I'm trying 2024.11.1 now. 

You may try clicking the three dots and selecting "Clean Build Files" and then attempt again.  Or restart your ESPHome container
[image]

Ragingfire105, 2024-11-23:
Followed both suggestions.  No change unfortunately.

Thank you for the rapid response.  I am quite a novice in the esphome space.  I only just recently started a small custom project of my own.  

Between my first post and now I updated HA to the latest so I'm assuming I also did the last one too.

MallocArray, 2024-11-23:
I know another user posted a similar issue several months ago and I think it worked itself out later, so might have to just wait for another release of ESPHome to come out.

Ragingfire105, 2024-11-24:
We shall see.  I tried reinstalling the device compiler but no change unfortunately. 

Thank you for looking into it.

MallocArray, 2024-11-25:
You could also try opening an issue with the ESPHome project itself and see if they have additional ideas:
https://github.com/esphome/issues

MallocArray, 2024-11-25:
Actually, searching the Issues for the last line of your original error came up with a result:

https://github.com/esphome/issues/issues/6467

Looks like they addressed the issue 6 hours ago so should be included in a future release of ESPHome

closed 2025-01-10 by MallocArray