MallocArray/airgradient_esphome #106: AirGradient Pro drops/loses connection from time to time

DariBer, 2025-01-10:
I have 3 AirGradient Pro running ESPHome code from this Github.

In my Home Assistant Dashboard, I can see that my units lose the connection from time to time, and I can't see what the problem is.

I have other ESPHome devices and I don't have these drops. So it's not my wifi or my HA installation.

This is how my code looks like

# AirGradient Pro V3.3 - V4.2
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  name: "airgradient-pro2"
  friendly_name: "AirGradient-Pro2"
  name_add_mac_suffix: "false"  # Must have quotes around value

# Enable Home Assistant API
api:

http_request:
   verify_ssl: "false"
   
ota:
  platform: esphome
  password: !secret ota_pwd

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  use_address: 192.168.25.7
  reboot_timeout: 15min

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-pro.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_d1_mini_board.yaml
  captive_portal: github://MallocArray/airgradient_esphome/packages/captive_portal.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  temp_humidity: github://MallocArray/airgradient_esphome/packages/sensor_sht40.yaml
  # temp_humidity: github://MallocArray/airgradient_esphome/packages/sensor_sht30.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  display: github://MallocArray/airgradient_esphome/packages/display_sh1106_single_page.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_d1_mini.yaml
  config_button: github://MallocArray/airgradient_esphome/packages/config_button.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml

binary_sensor:
  - id: !extend config_button
    pin:
      number: D7

display: 
  - id: !extend oled_display 
    rotation: 180°

MallocArray, 2025-01-10:
When you say it loses connection, are you observing the Uptime reset in HomeAssistant, or are you actually seeing gaps in your graphs?

I've always had issues with the devices randomly rebooting, especially with the ESP8266 based devices.  Often it seems to be related to the API calls to AirGradient taking too long to reply and causes the ESPHome watchdog to reboot the device.  But I don't see gaps in my graph when this happens, but I do see the Uptime reset.  

If you aren't using the AirGradient Dashboard, you could comment out the `airgradient_api` line and see if that helps stability.

Could you also share what firmware version you are running? You can see this when the device boots up, or in HomeAssistant, under Settings>Devices>ESPHome and select your AirGradient.  This will show which config version you are using along with what ESPHome version was used to compile it.
[image]

DariBer, 2025-01-10:
Might be it reboots as all sensors stops to work and I can't ping the device. 

I tried to disable the Gradient API as you wrote. 

See attached screenshot for version numbers. 

[image]

DariBer, 2025-01-10:
Looks like disable the AirGradient API did the trick. 

I haven't lost the connection since I disabled it and installed the new version.

MallocArray, 2025-01-10:
It could also be related to the API calls using HTTPS which needs even more memory on the constrained esp8266.  If you still want the API, you can try changing to `airgradient_api_d1_mini_insecure.yaml` to switch back to the insecure communication which may stabilize things.  I have one of my 3 Pro/Basic devices that I have to use the insecure one, while the other 2 work with the normal one, which doesn't make much sense.